OpenVSP/OpenVSP #31: Complete quoting for parameters of some CMake commands

elfring, 2015-03-11:
Some parameters (like "[${VSP_BINARY_DIR}/main.h](https://github.com/OpenVSP/OpenVSP/blob/588ac3dae3d3dfaaf3061ef9039025ef83fe2ee8/src/CMakeLists.txt#L54)" and "[${CMAKE_BINARY_DIR}/VSP_Libraries_Config.cmake](https://github.com/OpenVSP/OpenVSP/blob/73d3e0f2e4326323036a6c7a3f63300882d99f91/Libraries/CMakeLists.txt#L48)") are passed to CMake commands in your build scripts without enclosing them by quotation marks. I see that these places will result in build difficulties if the contents of the used variables will contain special characters like semicolons.

I would recommend to apply [advices from a wiki article](http://cmake.org/Wiki/CMake/Language_Syntax#CMake_splits_arguments_unless_you_use_quotation_marks_or_escapes. "CMake splits arguments unless you use quotation marks or escapes.").

ramcdona, 2015-08-27:
Both of these cases (and most of the CMake variables we use) are path names set automatically by CMake (not user provided).  As such, I think they use escape characters internally for spaces -- and I don't know of a file system that would allow a ; in a pathname.

Do you see a specific problem with these (or others), or do you simply suggest that every CMake variable be fully quoted every place it appears?

elfring, 2015-08-28:
I suggest generally to prepare also build scripts for [the handling of special characters](http://www.cmake.org/cmake/help/v3.3/manual/cmake-language.7.html#command-arguments "Description for CMake argument types").

ramcdona, 2023-03-12:
I'm closing this as it is very old and I'm not sure what action is actually being suggested.

If you feel strongly that this should be taken up, please start a discussion on the OpenVSP Google Group to explain what you're thinking.

If you have a specific pull request with justification, please push up a branch and issue a PR.

closed 2023-03-12 by ramcdona