OpenVSP/OpenVSP #60: VSPAERO -setup doesn't gracefully handle line or ellipse sub-surface types in DegenGeom

nbrakeES, 2016-09-01:
The creation of the setup template file with degenerate geom files containing sub-surfaces other than rectangles and control surfaces are not gracefully handled.  Execution of "vspaero.exe -setup []_DegenGeom" can result in a bad setup or program hang.

ramcdona, 2019-11-11:
@justingravett, can you look into this & fix or close?

justingravett, 2019-11-14:
Failed to reproduce on v3.19.0:

1. Add a wing
2. Add an ellipse sub-surface
3. Add a line sub-surface and place at U=0.38
4. Select Analysis->DegenGeom and press execute
5. Execute "vspaero.exe -setup []_DegenGeom" through the command line interface
6. VSPAERO setup file is generated successfully

Issue is closed.

closed 2019-11-14 by ramcdona