OpenVSP/OpenVSP #62: VSP crash on DegenGeom execution v3.9.0

nbrakeES, 2016-09-01:
To repeat 
1. open vsp
2. load the attached file
3. select Analysis->DegenGeom
4. press execute

This error is output to the command line and OpenVSP exits:

Internal error in segmentintersection():
  Topological inconsistency after splitting a segment.
  Please report this bug to jrs@cs.berkeley.edu
  Include the message above, your input data set, and the exact
    command line you used to run Triangle.

[test_multisurface_control_v2.zip](https://github.com/OpenVSP/OpenVSP/files/450919/test_multisurface_control_v2.zip)

ramcdona, 2016-09-05:
These errors are usually caused by true degenerate triangle edges in a CompGeom-type intersection.  Here it could be DegenGeom's call to CompGeom or MassProp.

You might check for control surface edges exactly aligning with one another -- or with the underlying mesh.

This file does not fail on my Mac.

ramcdona, 2019-11-11:
@justingravett, please check into this and fix or close.

justingravett, 2019-11-14:
Unable to reproduce in OpenVSP 3.19.0 Win32 or Win64. Issue is closed.

closed 2019-11-14 by ramcdona