OpenVSP/OpenVSP #65: VSPAERO solution dependent on model scale

nbrakeES, 2016-09-22:
[Google group issue](https://groups.google.com/d/msg/openvsp/zGGiektEQoY/S6TDTgUlFgAJ)

There is a scale dependence on the VSPAERO solution, small scale geometry returns invalid results.  

Procedure to replicate:
1. Open OpenVSP Version: 3.9.1
2. Add default wing
3. Open VSPAERO analysis GUI (Analysis --> VSPAERO)
4. Select reference lengths "From Model" (optional step)
5. Click "Generate Geometry"
6. Click "Create New Setup"
7. Click "Launch Solver" --> observe GOOD lift distribution in "Results Manager - VSPAERO" GUI
8. Scale wing by 0.001 (Wing --> Scale = 0.001 --> Accept)
9. Return to VSPAERO analysis GUI
10. Click "Generate Geometry"
11. Click "Create New Setup"
12. Click "Launch Solver" --> observe BAD lift distribution in "Results Manager - VSPAERO" GUI

ramcdona, 2019-11-11:
@justingravett, please check into this and fix or close.

justingravett, 2019-11-14:
Different issue in 3.19.0 Win32 and Win64. The default wing with scale set to 0.001 will cause OpenVSP to crash after selecting Analysis->DegenGeom...->Execute.

ramcdona, 2023-03-12:
This may still be an issue.  This would be a good one for you to have your team check out and build some unit tests around.  In particular, ramp up/down the scale gradually to see when/where failure occurs.

justingravett, 2023-03-15:
Hi Rob, I just checked the 0.001 scaling of the default wing geom with v3.32.1. While DegenGeom and VSPAERO run fine (lift distribution is symmetric), I do see numerous "triangle_create_mesh Error" reports in the console that are not present with the default scale wing. 

Maybe this is a change that I haven't kept up with, but I'm not able to run the solver more than once with either the default or 0.001 scaled wing. The "Launch Solver" button remains deactivated while the "Kill Solver" does not appear to do anything. My PC shows a substantial CPU load and I waited several minutes for the VSPAERO solver to become available again. The rest of OpenVSP works fine. Please let me know if this is a misunderstanding on my part or if you'd like me to make a separate issue (apologies if it's been brought up already).

closed 2025-07-19 by ramcdona