OpenVSP/OpenVSP #67: Incorrect load distribution in VSPAERO result at Wing Tip

nbrakeES, 2016-11-15:
Weird Wing Tip Phenomenon

nbrakeES, 2016-11-17:
Updated the procedure above with setting the airfoil to a cambered airfoil.  This is key to replicating the behavior results.

ramcdona, 2017-07-22:
Please try this with VSPAERO 4.1 included in OpenVSP 3.13.0

ramcdona, 2019-11-11:
@justingravett, please look into this and fix or close.

justingravett, 2019-11-14:
Different issue in OpenVSP 3.19.0 Win32 and Win64. The attached model will cause OpenVSP to crash when selecting Analysis->DegenGeom...->Execute.

[Hershey.zip](https://github.com/OpenVSP/OpenVSP/files/3848044/Hershey.zip)

ramcdona, 2023-03-12:
Another case you might want to have your team build a test case around.  Expect different results for thick vs. thin geometry.

justingravett, 2023-03-15:
Using the original geometry from the Google Group post, I'm still seeing the same odd spikes near the wing tips. The Hershey bar example Nick described and I shared a model of looks good in v3.32.1 with no wing tip issues.

closed 2025-07-19 by ramcdona