OpenVSP/OpenVSP #68: Whish: Geometry object for blueprints

Damov, 2016-12-30:
Hello everyone,

I like this software very much. Nevertheless, there is no geometry object like a flat plane, where one can put a texture on it of a blueprint to model for example the geometry of an aircraft. Is it possible to include in the next version a flat plane extra for blueprints?

Kind regards,
Daniel

ramcdona, 2016-12-30:
I'm not exactly sure what you're asking for.

However, have you tried setting a background image?  Under the 'Window' menu, 'Background', and then a button in the middle to select image.

Damov, 2016-12-31:
Hey, thanks for your answer. Yes, but if you rotate or zoom the window, the background image stays the same. In Rhino 3D they had a flat plane as geometry object, where you are able to put a texture of a blueprint.

Symbian9, 2017-04-08:
@Damov, starting from OpenVSP 3.11.0 there is "SVG 2D export" that give possibility to make 3view blueprints.
* https://github.com/OpenVSP/OpenVSP/releases/OpenVSP_3.11.0

But, as for me, it still need many improvements.

ramcdona, 2019-11-11:
You can actually add a square as a geometry, add a texture to it, and achieve this effect.

iforce2d, 2021-05-01:
Hi Rob
Could you elaborate on that procedure a bit more? Usually two hours of googling and twiddling around is enough for me to figure out how to do things, but not this time.
I tried using a box with textures, the texture becomes warped.
I tried importing a flat square .stl file, nothing happens (because not solid shape?)
I tried using planar slice to get a flat "meshgeom", but seems like textures don't apply to those.

ramcdona, 2021-05-01:
OpenVSP isn't really set up for this -- you have to trick it into working.

Add a Stack component.  Delete all the XSecs until you only have two left.  Make them both rounded rectangles with zero corner radius.

Set the width of both rectangles to something very small in your units.  Say .01.  Set the height to the dimensional height of your image - say 10.  Then, set the deltaX between the two cross sections to the length of your image, say 10.

Then, map the texture onto the object.  You'll need to scale and shift the image to get it where you want, but you should be able to figure it out from there.

"One of these days" we'll add proper support for this into OpenVSP -- making it 1000x easier to work with.  However, until that happens, this should get you started.

iforce2d, 2021-05-01:
Thanks, that works great. 
Yes, tricky to set up, glad I asked for help instead of continuing to try and figure it out myself :)

ramcdona, 2024-05-12:
Check out 3.39.0

closed 2024-05-12 by ramcdona