OpenVSP/OpenVSP #72: Export to AC3D (.ac) format

Symbian9, 2017-01-31:
Add possibility for export 3D mesh to AC3D format. It support assembly with parts  in single
* https://www.inivis.com/ac3d/man/ac3dfileformat.html  
## References
* https://github.com/NikolaiVChr/Blender-AC3D  
## Additionaly
* https://github.com/jcmellado/hangar
* https://github.com/epatel/ac3dreader
* http://code.j3d.org/impl/ac3d_loader.html

ramcdona, 2017-01-31:
What are you trying to accomplish with the AC3D file?  It looks like its uses are very similar to those we try to cover with X3D files.  Have you tried one of those?

This is a well formed issue (though possibly short on background like the above) with lots of great links, but I don't always check the issues here very often.  If you want to discuss a suggestion, most of that activity happens on the Google Group.

ramcdona, 2023-03-12:
The desire for an easy path from OpenVSP to various flight simulation programs is understood.  So far, none of the sponsors that fund OpenVSP development have made it a priority.

If someone from the community wants to work in this direction, I will be happy to help them out.  In the meantime, I'm closing this issue as it doesn't seem to be going anywhere.

As usual, general discussions are best held in the OpenVSP Google Group -- reserving this list as a place for specific bugs with reproducible steps.

closed 2023-03-12 by ramcdona