OpenVSP/OpenVSP #73: Crash on launch in Windows 10 (Version 1607, Build 14393.693)

jh502, 2017-02-03:
The VS2015 Debugger reports the following error:- 
Unhandled exception at 0x5EAA2481 (atigktxx.dll) in vsp.exe: 0xC000041D: An unhandled exception was encountered during a user callback.

Continuing execution gives
Exception thrown at 0x5EAA2481 (atigktxx.dll) in vsp.exe: 0xC0000005: Access violation reading location 0x00000000.

ramcdona, 2017-02-04:
My guess would be a problem with your graphics driver for this computer.  Have you installed a graphics driver specific to your chip?  Or does this computer just have the default Microsoft driver installed?

On Windows 10, OpenVSP requires a current OpenGL supporting graphics driver be installed.

jh502, 2017-02-04:
I'm currently using an R9 380 with standard Radeon drivers; I'm not aware of issues with other OpenGL applications, but I'll quickly update / verify and see if that has any effect.

jh502, 2017-02-04:
Good guess; An installation of new graphics drivers has me up and running.
For documentation sake, I'm now using the following:

Radeon Settings Version - 2017.0125.1845.33722
Driver Packaging Version - 16.50.2711-170125a-310791E
Provider - Advanced Micro Devices, Inc.
2D Driver Version - 8.1.1.1583
Direct3D® Version - 9.14.10.01235
OpenGL® Version - 6.14.10.13467
OpenCL™ Version - 21.19.413.0
AMD Mantle Version - 9.1.10.0165
AMD Mantle API Version - 102400
AMD Audio Driver Version - 10.0.1.1
Vulkan™ Driver Version - 1.4.1
Vulkan™ API Version - 1.0.37

Many thanks @ramcdona

closed 2017-02-04 by jh502