OpenVSP/OpenVSP #74: Build fails first time, works in second run

fernape, 2017-03-11:
Tested on FreeBSD. OpenVSP 3.10 works fine. With 3.11, the first time make is run, the compilation fails with this message:
`[ 54%] Building CXX object src/util/CMakeFiles/util.dir/SuperEllipse.cpp.o
<built-in>: In function 'float abs(float)':
<built-in>: warning: conflicts with previous declaration here [-Wattributes]
<built-in>: In function 'float abs(float)':
<built-in>: warning: conflicts with previous declaration here [-Wattributes]
[ 55%] Building CXX object src/util/CMakeFiles/util.dir/SVGUtil.cpp.o
In file included from /usr/home/fernape/redports.org/fernape/cad/openvsp/work/OpenVSP-OpenVSP_3.11.0/src/util/SVGUtil.cpp:11:0:
/usr/home/fernape/redports.org/fernape/cad/openvsp/work/OpenVSP-OpenVSP_3.11.0/src/util/SVGUtil.h:17:21: fatal error: XmlUtil.h: No such file or directory
 #include "XmlUtil.h"
                     ^
compilation terminated.
--- src/util/CMakeFiles/util.dir/SVGUtil.cpp.o ---
*** [src/util/CMakeFiles/util.dir/SVGUtil.cpp.o] Error code 1
`
Running make again without any modifications makes OpenVSP build fine.

The dependency of XmlUtil.h was introduced in 3.11 and I can see the variable XMLVSP_INCLUDE_DIR is properly set and I can not spot the problem so I can not provide a patch.

Any ideas?

ramcdona, 2017-03-11:
There was a circular dependency between src/util and src/xmlvsp.  I moved SVGUtil to geom_core in 697b3fa676bfb8.  Try the tip of master or https://github.com/OpenVSP/OpenVSP/tree/BuildFixes rather than the tagged 3.11.0.

closed 2017-03-11 by ramcdona

fernape, 2017-03-12:
> There was a circular dependency between src/util and src/xmlvsp. I moved
> SVGUtil to geom_core in 697b3fa
> Try the tip of master or https://github.com/OpenVSP/
> OpenVSP/tree/BuildFixes rather than the tagged 3.11.0.

Got it.

Thanks!