OpenVSP/OpenVSP #84: Spaces Needed with Mass Prop .txt file

RigelFive, 2017-07-25:
I have numerical outputs where the inertias for each axis are running together.  Example:

123456789012.23457692271.29123

It should be an easy fix to add a minimum number of spaces between numerical floating point values.

ramcdona, 2017-07-26:
What version of OpenVSP are you running?  Can you post a complete file showing this behavior -- seeing which line it occurs on will help narrow down the problem.

Looking over the code that prints out the mass prop file, I don't see where this could happen.  So more clues will help track this down.

ramcdona, 2019-11-11:
Please test with a recent version (say 3.19.0) and re-open if the problem persists.

closed 2019-11-11 by ramcdona