OpenVSP/OpenVSP #85: VLM not working in Version 3.13.0

AlwinW, 2017-07-30:
Using the windows complied version for 3.13.0 from [the official site](http://www.openvsp.org/download.php), VLM does not work in VSPAERO. There is no indication that VSPAERO is working (low CPU usage and no scrolling text in the updated "solver console". From the release notes of 3.13.0 it mentions "Update VSPAERO integration to support v4.1" so perhaps there would be a good start?
Otherwise, the new GUI for VSPAERO and parasitic drag looks very good!

AlwinW, 2017-08-03:
3.13.1 appears to have solved this problem. One point of interest though is a Mach number sweep that includes Mach = 1 (e.g. M = 1, 2, 3) will cause VSPAERO to crash. This could possibly due to asymptotic corrections based on Mach number.  I don't know if this was an existing issue or a new one introduced with VSPAERO 4.1

ramcdona, 2019-11-11:
Please test with version 3.19.0 and re-open if the problem persists.

closed 2019-11-11 by ramcdona