OpenVSP/OpenVSP #86: wing control_surf subsurface visual problem v3.13.0

marcovanderbijl, 2017-08-03:
wing control_surf subsurface visual problem v3.13.0
see attached screenshot of issue
the control subsurface lengths are all at 33%
The StartU is correctly drawn; the EndU has a reversal problem

thanks for programming all this!
[image]

marcovanderbijl, 2017-08-06:
Enforcing the 'surface end angles' solves the problem. So it seem so just be a little buggy initialization when adding the sub-surf.

ramcdona, 2017-08-06:
It is a bit hard to tell from this image what is going on -- I suspect you have multiple wings that you have abutted to make your overall shape.  That isn't a problem, but it is a little harder to tell what is going on.

You have the start/end U to 0,1.  0 actually starts down the end of a wingtip, and 1 is the end of the opposite wingtip.  If this is a single segment wing, you probably want your control surface to run from 0.33333 to 0.666666.  Please try this and see if you still observe the same issue.

marcovanderbijl, 2017-08-08:
The wing in the picture consists of 2 parts, the center section is highlighted red.
The outboard is in black, and is offset in x,y,z to attach to the outside of the center section.
Thus the highlight controllsurf is from startU 0 to endU 1, running along the whole trailing edge of the center section. This center section has 5 ribs, rib0 and rib1 are doubled for strength and each rib is different in length; the trailing edge is curved.
The non highlighted outboard section also has a controllsurf along the whole of the curved TE.

The controlsurf axles are touching correctly at the edge of  center and outboard wingsection as they are all set at 33% 
However the Upper/Lower shell breakline is incorrect. The start of the breakline is good, with the breakline opening towards the trailing edge. The outside (@ endU) of the breakline is drawn incorrectly, the breakline opens towards the leading edge.

Hopes this helps,
and I don't mind helping more.

ramcdona, 2017-08-08:
Ok, all the ribs, doubling, other wing parts, etc don't matter for this issue.  You should see the exact same behavior if you delete all other Geom's except for the wing with the surface that has this issue.

Next, please try my suggestion to change the start/end U to 0.333 and 0.666.

marcovanderbijl, 2017-08-10:
Thanks!
Setting the start/end U to other values than 0 or 1 and playing with the surf end angle I can sort of get to what I want. The rest of the visual oddness that I now have left over in my model is explained by how I build up the ribs.
Not  a bug,..... I should not have doubted the programming gods.
Closing issue.

closed 2017-08-10 by marcovanderbijl

ramcdona, 2017-08-10:
No programming gods here -- OpenVSP has lots of bugs and we need all the help we can get to fix them.

You shouldn't have to play around too much.

Let me try to explain what is going on...

A single-segment wing usually actually has three segments.  The root wingtip, the wing surface, and then the tip wingtip.  Under the hood, the U-parameter increments by 1 for each of these, so the parameter runs 0-1-2-3.  Then, to present the user with a parameter that always runs [0,1], we divide by 3.  Consequently, the root cap is [0,.333333], the main wing surface is [.33333,.66666], and the tip cap is [.66666,1.0].

A two segment wing has four segments [0 .25 .5 .75 1], etc.  If you set the Root/Tip Cap Type (under Plan tab) to 'None', you can eliminate those surfaces and the parameter will run [0,1] along the surface as you might have expected.  But then you don't get a closed wing surface...

When you specify a control surface chord length, (as x/c or x), OpenVSP must execute an iterative solver to figure out which W parameter is required to match the length you've requested.  This iterative solver gets confused when you put Ustart/Uend = 0,1 because the absolute 0,1 sections are a perfect straight line with the top/bottom surfaces coincident.  So, when the solver converges on the wrong surface, you get the strange behavior you've been seeing.

Setting Ustart/End to [0.3333,0.6666] for the control surface will actually cause it to start on the wing surface where you want it to.  (Alternately, you could turn off the end cap treatment).

When you choose to set the control surface angle, OpenVSP has to run a different solver that varies U,W of the control surface corner until it can match the desired length and angle.  Since that solver operates differently, it gets off the U=0 singularity and can find a more appropriate solution.

I hope this helps explain what is happening under the hood.

marcovanderbijl, 2017-08-10:
yes, it does. 
and turning off the rootcap of the wing fixed all the rest of the wrongness that I had.
thank you again.