OpenVSP/OpenVSP #87: Issues build under Ubuntu 16.04

AlexandreBorowczyk, 2017-08-12:
I've followed

marcovanderbijl, 2017-08-17:
Hi!

use the method that builds the libraries first.
copy the fluid executable from
'/buildlibs/FLTK-prefix/bin'  into
'/repo/src/vsp_aero/viewer' and '/repo/src/fltk_screens'
to get over the fluid problem

then build the openvspexe and end with a
$ make package
use the packaged build to run the program from.
using packagebuild makes openvsp find the scripts for podman etc.

in the hope it helps and with best regards,
M.

AlexandreBorowczyk, 2017-08-17:
Awesome it work perfectly!
Thanks for the help!

closed 2017-08-17 by AlexandreBorowczyk

marcovanderbijl, 2018-06-23:
I still use this thread (and the OP link to Ubuntu 16.04 build instructions) as my own build instruction manual.
I Just switched from Ubuntu16.04 to Mint18Sylvia (based on Ubuntu 16.04) and am
just leaving a breadcrumb for myself for next time:

add
python-dev
to the required dependencies list.
and also
sudo copy fluid executable to /usr/bin/