OpenVSP/OpenVSP #88: vspslicer does not start on Mac OS 10.11 and 10.12

RigelFive, 2017-09-20:
I've been excited to try out the latest in VSPAero 4.2.   So far the leading edge suction toggle seems to generate a different prediction than without on a nacelle duct lip.  It looks more reasonable.

When I start the vspslicer routine on a Mac OS 10.12, it goes directly to a Segmentation Fault: 11 error. 

I'm not planning to explore compiling after having tried a few times in the past.  But is it reasonable to try to compile vspslicer on a Mac... or just fix to have it distributed to work for Mac OS 10.12?  I have also tried on Mac OS 10.11.3 and get a similar response.

ramcdona, 2017-09-20:
Unfortunately, the slicer program is totally undocumented.

You must create a casename.cuts file that matches goes with the case you want to slice.   I'll give an example at the end of this note.

The *.cuts file contains the number of slices, and then a description of the cutting planes -- right now, it can only cut along x,y,z planes.

Then, you run slicer and pass the case name

./vspslicer casename

Slicer should compile as a part of the whole build -- VSPAERO (including slicer) is generally easier to build than OpenVSP, but it isn't trivial to just separate it out and just build that part.  Perhaps I should work on adding that as a build option.

Example casename.cuts follows:
2
y 0.0
y 1.0

closed 2017-09-20 by ramcdona

RigelFive, 2017-09-20:
Ok!  Got it to work.  Looks great.