OpenVSP/OpenVSP #89: Python API looking for VSPAERO in the wrong place

thomasleese, 2017-10-17:
When using the Python API, it is looking for the VSPAERO tools (`vspaero` and `vspviewer`) in the wrong place. With a default installation it is looking for them in: `/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS` because that is where the `python` binary is running from.

I believe the problem stems from these lines below where it is looking for the commands. Instead a better solution would be to simply use the system's `PATH` variable to find those commands which is a more standard way of finding executables.

https://github.com/OpenVSP/OpenVSP/blob/e953c566a99d7a8b5c31a691f8ac32442211a737/src/geom_core/Vehicle.cpp#L352-L359

thomasleese, 2017-10-17:
I should point out that this makes the Python API basically unusable without commenting out those lines of code.

ramcdona, 2017-10-23:
VSP expects the vspaero, vspviewer, and vspslicer (in the next version) executables to be in the same place as the vsp executable.  In the case of the Python API, it expects these to be in the same place as the _vsp.so binary file.

Searching the path isn't a good option because these executables change frequently -- with every update of OpenVSP.  So, it is best to co-locate the VSPAERO programs where OpenVSP itself is located.

If you aren't using VSPAERO, then these messages should be pretty harmless -- they will print out when the API is loaded.  From then, everything else in the API will work -- except for calling those binaries.  That seems to me to be a fairly reasonable situation.

closed 2017-11-25 by ramcdona

delonguevils, 2019-03-24:
@ramcdona I think there is still an issue linked to the call of the `PathToExe()` function from the python API. 

You wrote that:

> In the case of the Python API, it expects these to be in the same place as the _vsp.so binary file.

However, for the trials I made with the python API (OpenVSP_3.16.1 or OpenVSP_3.16.2, on windows and Linux computers), `PathToExe()` seems to return the **path to the python executable file**, instead of the path where _vsp.so is located. This means that, the python API searches (and try to run) vspaero, vspviewer, and vspslicer binaries in `C:\Anaconda2\` or `/usr/bin` (for example), instead of the location of the _vsp.so file, as you suggested.

This seems to be a bug specific to the python API, because running vspaero from the GUI of vsp leads to the expected behavior (m_ExePath  = ...`/OpenVSP/build/vsp`, which is the location of the vsp binary).

Would that not be preferable that, in case of use of the Python API, `PathToExe()` (or `m_ExePath`) would return/point to the location of _vsp.so or vsp.py ?