OpenVSP/OpenVSP #96: Build Failure for 3.15.0 - Missing links to xmlNewChild, xmlNewNode, etc.

fishforbreakfast, 2018-01-09:
Hi,
Just trying out 3.15.0.  I ran into this error when trying to compile:
[ 80%] Linking CXX executable apitest_g
../geom_core/libgeom_core.a(Airfoil.cpp.o): In function `FileAirfoil::EncodeXml(_xmlNode*&)':
Airfoil.cpp:(.text+0x6c0): undefined reference to `xmlNewChild'
../geom_core/libgeom_core.a(Airfoil.cpp.o): In function `CSTAirfoil::EncodeXml(_xmlNode*&)':
Airfoil.cpp:(.text+0x7fa): undefined reference to `xmlNewChild'
../geom_core/libgeom_core.a(BORGeom.cpp.o): In function `BORGeom::EncodeXml(_xmlNode*&)':
BORGeom.cpp:(.text+0x1bb0): undefined reference to `xmlNewChild'
../geom_core/libgeom_core.a(FeaStructure.cpp.o): In function `FeaStructure::EncodeXml(_xmlNode*&)':
FeaStructure.cpp:(.text+0x19496): undefined reference to `xmlNewChild'
FeaStructure.cpp:(.text+0x19550): undefined reference to `xmlNewChild'
../geom_core/libgeom_core.a(FeaStructure.cpp.o):FeaStructure.cpp:(.text+0x195ea): more undefined references to `xmlNewChild' follow
../geom_core/libgeom_core.a(XSec.cpp.o): In function `XSec::CopyFrom(XSec*)':
XSec.cpp:(.text+0xe2cf): undefined reference to `xmlNewNode'
XSec.cpp:(.text+0xe370): undefined reference to `xmlFreeNode'
../geom_core/libgeom_core.a(XSecCurve.cpp.o): In function `XSecCurve::CopyFrom(XSecCurve*)':

There are a lot more messages that I will have to try to capture and post here.  If this is a duplicate issue, my apology, and please disregard this issue.

Thank you & Regards,

Frederick

fishforbreakfast, 2018-01-09:
Here is a complete error list (see attached text file)
[3.15.0-xml-build_errors.txt](https://github.com/OpenVSP/OpenVSP/files/1615598/3.15.0-xml-build_errors.txt)

rreis, 2018-04-03:
Hi, did you solve this error? If yes, could you say how?

I am facing the same problem. And I have tried with both my system libxml2 and also with the libxml2 provided by OpenVSP

[ 81%] Linking CXX executable vspscript
../geom_core/libgeom_core.a(XSec.cpp.o): In function `XSec::EncodeXSec(_xmlNode*&)':
XSec.cpp:(.text+0x65a): undefined reference to `xmlNewChild'
../geom_core/libgeom_core.a(XSec.cpp.o): In function `XSec::EncodeXml(_xmlNode*&)':
XSec.cpp:(.text+0x11ca0): undefined reference to `xmlNewChild'
XSec.cpp:(.text+0x11ce6): undefined reference to `xmlNewChild'
../geom_core/libgeom_core.a(XSec.cpp.o): In function `XSec::CopyFrom(XSec*)':
XSec.cpp:(.text+0x11d8d): undefined reference to `xmlNewNode'

kind regards

fishforbreakfast, 2018-04-16:
Hi,

Still working on this.  I tried 3.16.1 with the same error.  I haven't
been able to find where to add the correct linker flags either.

I'll let you know if I find the error and what I had to do to make
things work.

Regards,

Frederick

On Tue,03 Apr 2018 11:40:49 -0700 Ricardo Reis <notifications@github.com>
wrote:

> Hi, did you solve this error? If yes, could you say how?
>
> I am facing the same problem. And I have tried with both my system
> libxml2 and also with the libxml2 provided by OpenVSP
>
> [ 81%] Linking CXX [...]

rreis, 2018-04-16:
Hi

I have solve this problem. Check the link bellow:

https://groups.google.com/forum/m/#!topic/openvsp/1dUUzURY1ko

Kind regards Ricardo

ramcdona, 2019-11-11:
Please test with version 3.19.0 and re-open if the problem persists.

closed 2019-11-11 by ramcdona