OpenVSP/OpenVSP #102: incorrect wing tip quarter chord in .fem file

peterrohl, 2018-05-22:
In V3.16.1, the wing tip quarter chord (QC) coordinates are incorrect in the .fem file. The values in the file are simply a copy of the wing tip leading edge coordinates (LE) instead of the QC coordinates.
   Planform:

   Root LE:   0.00000   0.00000   0.00000 
   Root TE:   4.00000   0.00000   0.00000 
   Root QC:   0.98026   0.00000  -0.00023 

   Tip LE:    5.19630   9.00000   0.00872 
   Tip TE:    6.19570   9.00000  -0.02617 
   Tip QC:    5.19630   9.00000   0.00872

ramcdona, 2019-11-11:
Please test with version 3.19.0 and re-open if the problem persists.

closed 2019-11-11 by ramcdona

peterrohl, 2019-11-12:
Still not working. Now the X and Y coordinates are incorrect for the entire wing tip
This is the same wing as before, straight-tapered, wing span of 18, no dihedral (so the tip should be at 9):
(generated with 3.19.0)
Planform:

   Root LE:   0.00000   0.00000   0.00000 
   Root TE:   4.00000   0.00000   0.00000 
   Root QC:   0.98026   0.00000   0.00000 

   Tip LE:    4.92267   8.52632   0.00000 
   Tip TE:    6.08057   8.52632   0.00000 
   Tip QC:    5.19241   8.52632   0.00000