OpenVSP/OpenVSP #111: Flatpak manifest

Murmele, 2018-10-01:
Hello,

I created a flatpak manifest for OpenVSP, but when I compile it, I get the following error:

    [ 47%] Building CXX object src/vsp_graphic/CMakeFiles/vsp_graphic.dir/src/VertexBuffer.cpp.o
    /usr/lib/gcc/x86_64-unknown-linux/6.2.0/../../../../x86_64-unknown-linux/bin/ld: ../../../../../../Libraries-prefix/src/Libraries-build/FLTK-prefix/lib/libfltk.a(Fl_Window_shape.cxx.o): undefined reference to symbol 'dlsym@@GLIBC_2.2.5'
    //lib/libdl.so.2: error adding symbols: DSO missing from command line
    collect2: error: ld returned 1 exit status
    make[5]: *** [src/vsp_aero/viewer/CMakeFiles/vspviewer.dir/build.make:583: src/vsp_aero/viewer/vspviewer] Error 1
    make[4]: *** [CMakeFiles/Makefile2:2509: src/vsp_aero/viewer/CMakeFiles/vspviewer.dir/all] Error 2

Do anybody know how I can solve one of this errors? 

Here you can find the manifest:
https://github.com/Murmele/OpenVSP/tree/Flatpak

You can compile it with (when flatpak-builder is installed):

     flatpak-builder --repo=meinRepo2 --force-clean openvsp org.openvsp.openvsp.json

ramcdona, 2019-11-11:
We now build an Ubuntu *.deb file.  If this does not meet your needs, please re-open (or make a new issue) with specifics about your needs.

closed 2019-11-11 by ramcdona

Murmele, 2019-11-24:
I'm using Arch linux, so I'm not able to install your package. A flatpak is independed on the linux distribution you are using.

acxz, 2019-12-22:
@Murmele there is now an [`openvsp`](https://aur.archlinux.org/packages/openvsp/) package on the AUR on ArchLinux.