OpenVSP/OpenVSP #113: Snap Package Support

mikeroyal, 2018-12-27:
I was thinking the install and setup could be improved for Ubuntu and other Linux distros by having a [snap package](https://snapcraft.io/) version. If it's not a priority would you mind if I made the snap as a contribution?

[Crafting your first snap](https://tutorials.ubuntu.com/tutorial/create-your-first-snap#0)

[Build and test snap](https://snapcraft.io/build)

[Getting Started snap template](https://github.com/snapcrafters/fork-and-rename-me)

ramcdona, 2018-12-29:
I am totally unfamiliar with Snap.  However, I would love to have help packaging OpenVSP for any variety of Linux -- I had assumed it would be for a some distribution's native package system, but I'm sure that any progress in this direction will help others along.

Let me know if you need any help.

ramcdona, 2019-11-11:
We now have Ubuntu *.deb files available.  If this does not meet your needs, please re-open or start a new issue that details your specific need.

closed 2019-11-11 by ramcdona