OpenVSP/OpenVSP #117: Mass Prop Calculation Issue Win10 - reopened

ge98fiy, 2019-02-27:
Hey Rob,
finally I found the time to continue working on this issue and I found that deleting the previous meshes does not help. The random part selection also happens when I load the complete aircraft and then do a mass prop calculation (without having computed the mass props before). Sometimes all parts are taken and sometimes not (see attached figures as examples - both are taken directly after the mass prop calculation of the newly loaded configuration). This time I used the newest release (3.16.2). Do you have an idea of why this is happening? Thanks for your help!
[image]
[image]

_Originally posted by @ge98fiy in https://github.com/OpenVSP/OpenVSP/issues/103#issuecomment-467855362_

ge98fiy, 2019-04-11:
Hey Rob, 
did you find the time to have a look at the problem? 
Thanks,
Anna

ramcdona, 2023-03-12:
If you are still having general problems, please post a question to the Google Group -- possibly including a test file.

If you have a specific bug with directions of how to exactly reproduce it, please open a new issue here.

closed 2023-03-12 by ramcdona