OpenVSP/OpenVSP #118: VSP crashes or only showes white background when launched in Windows 10 (Version 1803, Build 17134.590)

WhisperOfHeart, 2019-03-07:
This problem happens on two laptops.
The first laptop has the Intel(R) HD Graphics 5500 and NVIDIA GeForce 940M installed. 
With Intel(R) HD Graphics 5500, OpenVSP only shows a white background when launched. All the components can be added, but can not be shown.
With NVIDIA GeForce 940M, OpenVSP directly crashes.

The second laptop has Intel(R) HD Graphics 620 installed.  OpenVSP directly crashes when launched.

I tried 11 versions(3.16.2 3.15.0 3.14.2 3.13.3 3.12.1 3.9.1...) of OpenVSP, the problem is same,

I find this problem may be caused by OpenGL. But with the current OpenGL supporting graphics driver be installed,  the problem is still there.

By the way, OpenVSP works fine on a laptop with NVIDIA GeForce GT 745M and Intel GMA HD 4600 installed.

ramcdona, 2023-03-12:
There is nothing the developers can do to resolve problems like this without access to the computer in question.

closed 2023-03-12 by ramcdona

jeffallen, 2023-09-26:
We just saw this problem with an NVIDIA Quadro P2200 card. The problem turned out to be that we were using the Microsoft built-in driver, and not the Nvidia one. The solution was to use Windows Update, and to click the extra "please install video drivers" box.

I know it is very difficult to debug OpenGL problems, but it would certainly be nicer if OpenVSP could exit with a message on the console of, "Cannot initialize OpenGL surface, please try updating your drivers" or something. I was not able to run a debug version on the affected system in order to find out where the crash happens.

commit 5844ccb1 (2023-10-28): Better error message when OpenGL is broken

Fixes #118