OpenVSP/OpenVSP #124: Geometry Error Causes VSP to Crash

EmileeWeir, 2019-04-11:
This just started happening a couple of days ago - I initially thought it was an issue with my external hard drive (I keep dropping it, causing it to unplug while running) and was originally not going to worry about it, but it just now happened to a model I started and saved directly to my laptop's internal hard drive. I need the vertical tail section to be a certain area and am now finding that, as I change the Section Planform geometries and, it will randomly change all of the non-selected planform geometries to "nan(undefined)". When this happens, if I attempt to change any of the Section Planform options for that particular geometry, it crashes the program. It will continue to do this every time I open the model, and eventually, the model cannot be opened. It will then return an error message when I try to open the model: "Failure in SkinRibs set_conditions". I was originally running version 3.16.1-win32 where the issue started. This continues to happen with the latest version. I have no idea why VSP is doing this and it has never happened before, but I suspect it could be related to the latest Windows updates (disclaimer: I am by no means even an amateur with computers but this is the only cause I can think of).

EmileeWeir, 2019-06-11:
Update: Yesterday, VSPAero also stopped working when trying to calculate models at 0 degrees AOA:

[image]

VSPAero is also producing weird results when other angles are run:

[image]
[image]
[image]
[image]

I tried having Microsoft Support reinstalling an older version of Microsoft (should have been the update before all of this occurred), as it is also affecting my ability to run Microsoft SQL. VSPAero still will not run models at a 0 degree AOA, even after deleting and reinstalling OpenVSP.

EmileeWeir, 2019-07-08:
Update: I'm still having issues with it corrupting files, but now I cannot even get VSPAero to run! It crashes every time I try to run the program on a file I just created a few days ago (though strangely it seems to still work with older models). I have tried contacting Microsoft Support about this four times now (the first time my Windows 10 version was restored to a previous version, the second time Windows 10 updated, the third ended up getting disconnected, and the fourth suggested contacting OpenVSP support while they submitted a report to Windows 10 engineers) and nothing thus far has provided a more than temporary fix. I need to be able to run VSPAero on this new file and am hoping that it does not corrupt!

[image]

ramcdona, 2023-03-12:
If you are still having this trouble, please post a message to the Google Group.  Possibly post your *.vsp3 model if you can.

If you have a specific bug to report, please open a new issue with a reproducible test case.

closed 2023-03-12 by ramcdona