OpenVSP/OpenVSP #127: Arch Linux Package for OpenVSP

acxz, 2019-07-25:
Hello @ramcdona, I finally got around to fixing up the Arch Linux package for OpenVSP https://aur.archlinux.org/packages/openvsp/. I was wondering if there is any way to add instructions for it on the install page of the documentation http://openvsp.org/wiki/doku.php?id=install.

ramcdona, 2019-11-11:
Send me an email and I am happy to give you edit permissions to the Wiki.

closed 2019-11-11 by ramcdona

acxz, 2019-12-22:
@ramcdona my apologies for responding late. I have found an email address to contact you though I am not sure if you will receive it (might be an older email account). Can you please respond back here, if you have not yet received an email from me, along with some way to contact/email you?

ramcdona, 2019-12-24:
If you pull down the OpenVSP repository, my email address is included in all the commit messages I make.  A long time ago, this was a calpoly.edu address -- probably what you had fail.  The best address for you to use is now my gmail address.

Rob

acxz, 2019-12-24:
Ahh yes I had tried your old calpoly email. Silly me, the commit log was the first thing I should have checked. Just sent the email.