OpenVSP/OpenVSP #130: Setup Travis CI for build AppImage (portable binary package for Linux)

Symbian9, 2019-09-09:
Lets setup Travis CI for auto-build *AppImage* (portable package for Linux) for each commit:

1. https://github.com/appimage/AppImageKit
1.1. https://github.com/probonopd/linuxdeployqt
1.2 https://github.com/linuxdeploy
2. https://github.com/probonopd/uploadtool

For example, take a look on *LeoCAD* `.travis.yml` implementation:

- https://github.com/leozide/leocad/blob/master/.travis.yml

And here is what resulted builds would look like:

- https://github.com/leozide/leocad/releases/tag/continuous

For any help with creating *AppImage* build scripts ask @probonopd and @TheAssassin.

TheAssassin, 2019-09-09:
Please stop pinging me in every of your issues. The best way to get in touch is to send people to our IRC channel. I'm happy to help, but this kind of spamming is not appreciated.

TheAssassin, 2019-09-09:
https://docs.appimage.org/user-guide/faq.html#question-where-do-i-get-support

Symbian9, 2019-09-10:
> Please stop pinging me

Sorry, in the future will link only FAQ page.

probonopd, 2019-09-10:
Same for me. I am also happy to help, but only _after_ the original author(s) of an application have expressed their interest to have, distribute, and support AppImage builds. In which case I may send a pull request doing the AppImage builds.

Symbian9, 2019-09-24:
As start point for solving this task here is build script for *Ubuntu 18.04* Linux distributive:

- https://github.com/marek-cel/openvsp-building-script

ramcdona, 2019-11-11:
OpenVSP now builds an Ubuntu *.deb file -- if this does not meet your needs, please open a new request with specifics about your need.

closed 2019-11-11 by ramcdona

TheAssassin, 2019-11-11:
> OpenVSP now builds an Ubuntu *.deb file -- if this does not meet your needs, please open a new request with specifics about your need.

That sentence makes little sense, given this issue *already* asks *specifically* about providing an AppImage...

ramcdona, 2019-11-11:
@TheAssassin If you check this thread, you wanted to be left out of this one (:.  This isn't really about AppImage.

OpenVSP is a program that has previously not been packaged for Linux in any way.  Users had to build it for themselves on any flavor of Linux.

We very recently automated creating a *.deb file for Ubuntu.  I expect this will take care of >90% of our Linux user's needs.

I was closing the original request for OpenVSP packaging as AppImage -- not because it is the same thing or the user wasn't specific enough -- but because I suspect that an Ubuntu *.deb will be 'good enough' for the creator of the issue.

If they come back asking for an AppImage, I would still like to understand why -- what is the underlying use case that they can't 1) build it themselves 2) use the Ubuntu *.deb or 3) the only solution to their problem is an AppImage 4) They can't build the AppImage themselves.

Symbian9, 2019-11-12:
> I suspect that an Ubuntu *.deb will be 'good enough' for the creator of the issue.

Could you at least create one more *.deb for just for Debian 9.x? This would good enough.