OpenVSP/OpenVSP #140: Crash when adding a prop as child of a fuselage

iforce2d, 2019-12-19:
http://www.iforce2d.net/tmp/plank250.vsp3

Open file
Select "FuselageGeom"
Select "PROP" in the geom browser
Click "Add"

iforce2d, 2019-12-19:
hmm..... wait a minute. Adding a prop causes a crash every single time, regardless of what was selected as the parent. It even crashes when adding a prop to a fresh empty file.

justingravett, 2020-01-08:
I tried replicating your issue in 3.20.0 Win64 and was able to add a prop to your plank250.vsp3 file without OpenVSP crashing. Can you please provide additional information regarding your version of OpenVSP and operating system?

iforce2d, 2020-01-08:
Sorry, looks like this was my fault.

I'm running on 64bit Linux (Fedora 30). I forked in August last year and changed a few minor things (line color and thickness, default axes for sweep graph in VSPAero results).
https://github.com/iforce2d/OpenVSP/network

One of my changes was to remove the secondary Y axis that was obscuring part of the VSPAero results graph, which I did by commenting out some stuff:
https://github.com/iforce2d/OpenVSP/commit/9b423ce81775f29b172f4382f50768f8ac299c84

I also reported this issue and it was later fixed properly in the original repo:
https://github.com/OpenVSP/OpenVSP/commit/4385534b0dfb6495fcdecb8fedc6644a2bf1e05a

I realized this just now when I attached gdb, which I should have done earlier, but didn't because the method to trigger the crash seemed totally unrelated to any changes I made. I mean what does this have to do with VSPAero, and why would the viewer work fine for geometries other than PROP:

Start OpenVSP
Select "PROP" in the geom browser
Click "Add"

Anyway, nothing needs fixing, sorry for the bother.

closed 2020-01-08 by iforce2d