OpenVSP/OpenVSP #141: Crash when running VSPAERO on fuselage with "ROUNDED RECTANGLE" cross-sections

iforce2d, 2019-12-19:
http://www.iforce2d.net/tmp/plank250-fuseOnly.vsp3

Open file
Open VSPAERO
Click "Launch solver"

Terminal output:

ERROR 7: Could not open Load file: /home/chris/projects/vsp/hangar/plank250/plank250-fuseOnly_DegenGeom.lod
	File: /home/chris/builds/OpenVSP/OpenVSP-iforce2d/src/geom_core/VSPAEROMgr.cpp 	Line:2008
WARNING: xDoubleData.size() and yDoubleData.size() must be equal AND greater than 0
	xDoubleData.size()=0
	yDoubleData.size()=0
	File: /home/chris/builds/OpenVSP/OpenVSP-iforce2d/src/gui_and_draw/VSPAEROPlotScreen.cpp 	Line:2282

If you change the ROUNDED RECTANGLE sections of the fuselage to something else, eg. ELLIPSE, the crash does not occur.

ramcdona, 2019-12-29:
This is somewhat a known issue.

This happens when OpenVSP smashes the geometry down to a VLM.  The parallel sides of a rounded rectangle collapse to degenerate panels that cause the crash in vspaero.  OpenVSP then crashes in turn.

From a VLM perspective, you can change those cross sections to ellipses and re-do the analysis, you will get a better result.  This isnt the easiest thing to work with from a modeling perspective, but there also isn't an obvious easy way for us to fix it...

If your geometry doesn't change too much, you can copy/paste your fuselage and have a duplicate version with the ellipses -- then you can use Sets to have a VLM set and another set for other purposes.

andrianu, 2020-01-03:
Hi,
indeed, have the same issue, but worked around it, by selecting GENERAL_FUSE, then edited TopStr and BotStr, to value of 1.5-2, and looks very similar to a rounded rectangle .

ramcdona, 2020-01-04:
That will work, but it is not the best idea.  When you DegenGeom the shape to form the VLM representation, the distribution of points will be poor.  Changing it to an ellipse will have a better point distribution and will have the same VLM shape.

closed 2020-01-04 by ramcdona

iforce2d, 2020-01-04:
fwiw I switched to using ellipse and there were no issues with that.