OpenVSP/OpenVSP #142: libGLEW version

cafinux, 2019-12-25:
I tried installing openVSP on Debian Buster, but hit a problem when it called libGLEW-2.0 but 2.1 is installed. The later Ubuntu's use 2.1 version too. Btw openVSP runs perfectly on VM I created to use it on, just a bit of a pain firing up VM to use it. Thanks everybody for the cool application!

acxz, 2020-01-26:
@cafinux Can you give us the specific error output? I am using openvsp with glew version 2.1.0 and it works just fine.

reece146, 2020-06-27:
I ran into this also on Ubuntu Budgie (20.04). Fresh OS install this AM. Most likely a case of 'versionitis' re: 18.04 vs 20.04 parsing component version numbers.

```
root@digit:~# dpkg -i OpenVSP-3.21.1-Ubuntu-18.04_amd64.deb 
Selecting previously unselected package openvsp.
(Reading database ... 245305 files and directories currently installed.)
Preparing to unpack OpenVSP-3.21.1-Ubuntu-18.04_amd64.deb ...
Unpacking openvsp (3.21.1) ...
dpkg: dependency problems prevent configuration of openvsp:
 openvsp depends on libglew2.0 (>= 1.12.0); however:
  Package libglew2.0 is not installed.
 openvsp depends on libpython3.6 (>= 3.6.5); however:
  Package libpython3.6 is not installed.

dpkg: error processing package openvsp (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 openvsp

root@digit:~# apt list --installed | grep libglew

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libglew-dev/focal,now 2.1.0-4 amd64 [installed]
libglew2.1/focal,now 2.1.0-4 amd64 [installed]

root@digit:~# apt list --installed | grep libpython

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libpython3-all-dev/focal,now 3.8.2-0ubuntu2 amd64 [installed]
libpython3-dev/focal,now 3.8.2-0ubuntu2 amd64 [installed]
libpython3-stdlib/focal,now 3.8.2-0ubuntu2 amd64 [installed,automatic]
libpython3.8-dev/focal-updates,focal-security,now 3.8.2-1ubuntu1.1 amd64 [installed,automatic]
libpython3.8-minimal/focal-updates,focal-security,now 3.8.2-1ubuntu1.1 amd64 [installed,automatic]
libpython3.8-stdlib/focal-updates,focal-security,now 3.8.2-1ubuntu1.1 amd64 [installed,automatic]
libpython3.8/focal-updates,focal-security,now 3.8.2-1ubuntu1.1 amd64 [installed]
root@digit:~# 

root@digit:~# lsb_release --all
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04 LTS
Release:	20.04
Codename:	focal

cibinjoseph, 2020-09-19:
I've tested OpenVSP for Ubuntu 20.04 and it seems to work without making any changes. However, if you use the Ubuntu 18.04 deb file from the Downloads page to install it, it will not work. This seems to be what's happening for you. We'll probably upload the deb file for Ubuntu 20.04 soon!

As of now, you'll need to compile OpenVSP using the instructions provided in the wiki or you can directly run this [build script](http://openvsp.org/wiki/doku.php?id=openvsp_build_script) on your system with default settings and it should give you the deb installation package. Make sure your system is updated too!

cibinjoseph, 2020-09-19:
Related to Issue #153 
This issue may be closed when an Ubuntu 20.04 deb package is provided in the Downloads page.
For generating the deb package, OpenVSP does not require any changes other than having to be compiled on Ubuntu 20.04

cibinjoseph, 2020-10-04:
The Ubuntu 20.04 deb package file is now available from the [downloads](http://openvsp.org/download.php) page and has been tested. 
This issue may be considered resolved.

ramcdona, 2023-03-12:
I'm closing this issue since the Ubuntu packages are now pre-built for download with each version of OpenVSP.

If you need to build for yourself and are still having trouble, please direct your questions at the OpenVSP Google Group.

closed 2023-03-12 by ramcdona