OpenVSP/OpenVSP #143: Generate planar slice without the original geometry?

iforce2d, 2020-01-08:
This is not really an issue, but I don't know of a better place to ask.

Over the last year I have been designing model planes in OpenVSP and building them from slices of insulation foam, using the planar slice feature to generate the contour for each slice. Thanks to being simulated in VSPAERO they have all flown pretty well, and I love the blended wing feature too. https://youtu.be/6xAY89Yqg-c?t=298

The workflow so far has been to use planar slice to generate slice geometry, export as STL to Blender where I separate the slices as individual objects, then run a Blender script to export these to a 2D CAD file (.dxf) and lay them out side by side.

In the early days I found the intermediate Blender stage handy to visualize where components (spar, battery etc) would fit, but recently I have been making these objects in OpenVSP so the Blender step is not really necessary for that now. Having an intermediate step also slows things down a lot when you need to do many many repeated exports while working on the design.

What I would like to do is write a script to convert the STL to .dxf directly and cut the turnaround time down to almost nothing. I have experience working with 3d meshes in general so I could find the outer contour of each slice from the unordered triangles, but the problem is that the STL also contains the original mesh that the slices were generated from, and some of those vertices are coincident with the slice vertices. If not for that, I could find which island is the original geometry (eg. by bounding box size) and ignore it.

So the question is, where in the OpenVSP code would I modify to not have the original mesh included in the geometry generated by planar slice? I had a look around myself and I get the impression it shouldn't be too hard to do this, but couldn't figure it out.

ramcdona, 2020-01-08:
The best place for this kind of question is the OpenVSP Google Group, available via the [OpenVSP website](http://openvsp.org/discuss.shtml), or the normal [Google Group page](https://groups.google.com/forum/#!forum/openvsp).

After you run Planar Slice, you will notice a new MeshGeom is created and added to your model.  It will typically appear at the bottom of the Geom Browser.  If you select that MeshGeom, a window will open to let you edit its properties.  There is an 'Other' tab there.  On the Other tab, there is an option to visualize the Mesh and/or the Slices.  Turn off the Mesh.

Then, when you export the STL, the Mesh will not be included in the file.

We have occasionally had requests for similar features from the modeling community.  Often, there is interest in adding features (spars, caps, D-boxes, etc) to wings and then slicing them before sending to a laser cutter for a built-up balsa wing.  Most of the heavy lifting for this use case (and yours) is already done by OpenVSP, it really would just take someone who fully understands the use cases to come up with an elegant UI and workflow for this sort of thing.

If you want to add your idea to OpenVSP itself, there are some bits and pieces laying around that may be helpful.  The code for 'Projected Area...' has the ability to extract the outline from a planar triangular mesh.  Some of this functionality comes from the 'clipper' library.

Also, there is already code for exporting DXF files.  It is meant to give a jump-start to someone who wants to make a 3-view drawing of their model in AutoCAD or other similar programs.  This isn't exactly what you want, but it would be a head start to the file format.  Finally, the 'outline' option in the DXF files uses the above Projected Area option to generate the outline and feed to the DXF file, so there is some overlap in code already.

closed 2020-01-08 by ramcdona

iforce2d, 2020-01-09:
Thanks Rob, that's perfect.