OpenVSP/OpenVSP #149: OpenVSP Skinning Algorithm

NaomiUnique, 2020-07-06:
Hello, I was wondering if anyone had figured out or found the algortihm that OpenVSP uses for skinning? And if so, would you mind showing me where it is?

ramcdona, 2020-07-10:
Was the response I left on the Google Group (that you didn't respond to) insufficient?

closed 2020-07-10 by ramcdona

NaomiUnique, 2020-07-10:
Hello Rob,

No, it was great. I was just looking for more information to better understand it, and my apologies, this was my first time using a forum like this, thank you for taking the time to respond to me.

Best,
Naomi U. Turner

Freshman Liaison, National Society of Black Engineers (B.O.S.S Chapter)
Electrical Engineering
Clarence M. Mitchell, Jr. School of Engineering
Morgan State University

Email: naomiuturner@gmail.com:

ramcdona, 2020-07-10:
Sorry to bust your chops a bit.  I am more than happy to elaborate on my answer or answer followup questions.  Or, perhaps I've misunderstood your question.  In any case, the Google Group is probably the best place for a back-and-forth.  Good job for finding it and asking there.

I realize my original answer is probably a bit frustrating.  There isn't a short/simple one-equation answer to your question.  That said, I bet there may be another answer that will work better for you.  Why are you asking?  Not being defensive, but truly what is your goal?  What are you trying to accomplish?  Are you just curious?  Do you want to duplicate that part of OpenVSP (if so, why, how, where)?

If you came in and said 'I think OpenVSP's skinning could apply to XYZ other problem, how does it work?', then we might be able to take a different path to answer your real question.

NaomiUnique, 2020-07-11:
I was just trying to get a look at the code to better general understand how the skinning is effected, but if I have a more specific question, I will definitely shoot your way!

Naomi U. Turner

Freshman Liaison, National Society of Black Engineers (B.O.S.S Chapter)
Electrical Engineering
Clarence M. Mitchell, Jr. School of Engineering
Morgan State University

Email: naomiuturner@gmail.com: