OpenVSP/OpenVSP #151: [Build] `SW_SHOWNORMAL` was not declared in this scope

acxz, 2020-07-28:
@ramcdona thanks for the new release!

I'm trying to update the Arch Linux package and I ran into a build error when compiling.

```
/home/acxz/vcs/git/github/acxz/pkgbuilds/openvsp/src/OpenVSP-OpenVSP_3.21.2/src/gui_and_draw/ParasiteDragScreen.cpp: In member function ‘virtual void ParasiteDragScreen::GuiDeviceCallBack(GuiDevice*)’:
/home/acxz/vcs/git/github/acxz/pkgbuilds/openvsp/src/OpenVSP-OpenVSP_3.21.2/src/gui_and_draw/ParasiteDragScreen.cpp:1194:35: error: ‘SW_SHOWNORMAL’ was not declared in this scope
 1194 |                       NULL, NULL, SW_SHOWNORMAL );
      |                                   ^~~~~~~~~~~~~
/home/acxz/vcs/git/github/acxz/pkgbuilds/openvsp/src/OpenVSP-OpenVSP_3.21.2/src/gui_and_draw/ParasiteDragScreen.cpp:1193:9: error: ‘ShellExecute’ was not declared in this scope
 1193 |         ShellExecute( NULL, "open", "http://www.openvsp.org/wiki/doku.php?id=parasitedrag",
      |         ^~~~~~~~~~~~
make[5]: *** [src/gui_and_draw/CMakeFiles/gui_and_draw.dir/build.make:602: src/gui_and_draw/CMakeFiles/gui_and_draw.dir/ParasiteDragScreen.cpp.o] Error 1
```

ramcdona, 2020-07-28:
Yep.  Had some problems trying to roll out the release last night.  If you look a couple commits past master, I've fixed this already.

acxz, 2020-07-28:
Can you release a new hotfix?

ramcdona, 2020-07-28:
Just use 9008c9ffa1b2c .

closed 2020-07-28 by ramcdona

acxz, 2020-07-28:
I'll direct users to use `openvsp-git` instead of the `openvsp` package on archlinux then.

ramcdona, 2020-07-28:
Your concept of a hotfix doesn't mean anything to me.

I am completely unfamiliar with the Archlinux distribution and packaging
system.

What would you do if you had to apply some patches that were not in
upstream?  Just apply 970d41dd53bd (the only one of the three changes that
matters for Linux) as a patch to your own build system.  Or, point your
build at a later commit, or do whatever you have to do.

Rob

acxz, 2020-07-28:
Yes I can patch the package locally, I'm testing the master and the commit to see if the issue is resolved on my end. Once I confirm it, I will apply the patch on the archlinux release.

I just thought it would make sense to have a working release for everyone. Sorry for being too forward about this.. (I should have tested master first before opening this issue)

openvspmachine, 2020-07-28:
master on OpenVSP/OpenVSP is always the latest stable release.  It gets tags when I make a release -- additional changes beyond that when I have problems with the release process (like this time).

All development happens in other forks, or at least in other branches.  Often, I keep branches called Fixes and Next.  Next is what I use to integrate all the various development branches before a release.

If your release system just tracks master on OpenVSP/OpenVSP, you should be fine.