OpenVSP/OpenVSP #153: Installation fails on Ubuntu 20.04 LTS with unmet dependencies

pmsagdeo, 2020-09-06:
Tried to install on Ubuntu.  Failed for the reason above.   Please fix, or advise on how to install dependencies.  Thanks. 
[VSPError09062020.pdf](https://github.com/OpenVSP/OpenVSP/files/5179718/VSPError09062020.pdf)

cibinjoseph, 2020-09-17:
The current version of OpenVSP was tested for Ubuntu 18.04. Are you using a different version?
Can you try installing with the apt package-manager using the command:
`sudo apt install ./OpenVSP-3.21.2-Ubuntu-18.04_amd64.deb`
If this throws an error. Post the output here. That should give us a hint of which dependencies you're missing.

Alternately, you can also try building OpenVSP using the instructions provided at the [wiki](http://openvsp.org/wiki/doku.php?id=ubuntu_instructions).

pmsagdeo, 2020-09-18:
I tried to install it on Ubuntu 20.04 LTS.  I will test the procedures below and let you know the outcome.  Thanks.

 

Pradip

 

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Virus-free.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

cibinjoseph, 2020-09-19:
I've tested OpenVSP for Ubuntu 20.04 and it seems to work without making any changes. However, if you use the Ubuntu 18.04 deb file in the Downloads page to install it, it will not work. We'll probably upload the deb file for Ubuntu 20.04 soon!

As of now, you'll need to compile OpenVSP using the instructions provided in the wiki or you can directly run this [build script](http://openvsp.org/wiki/doku.php?id=openvsp_build_script) on your system with default settings and it should give you the deb installation package. Make sure your system is updated too!

cibinjoseph, 2020-09-19:
Related to Issue #142 
This issue may be closed when an Ubuntu 20.04 deb package is provided in the Downloads page.
For generating the deb package, OpenVSP does not require any changes other than having to be compiled on Ubuntu 20.04

pmsagdeo, 2020-09-19:
Thank you.  I will compile and test.  Will let you know the outcome.  It may be a few days before I can do this, though.

 

Pradip 

 

This email has been checked for viruses by AVG.
https://www.avg.com

cibinjoseph, 2020-10-04:
The Ubuntu 20.04 download file is now available from the [downloads](http://openvsp.org/download.php) page and has been tested. 
This issue may be considered resolved.

closed 2020-10-21 by ramcdona