OpenVSP/OpenVSP #155: Segmentation fault when exporting STEP file

jrparks9469, 2020-09-15:
**Description**
When I attempt to export a STEP file of the default wing geometry, I receive a segmentation fault error. This happens for both trimmed and untrimmed STEP files. It doesn't happen for IGES file.

**OS details**
OS: Manjaro 20.1
Kernel: x86_64 Linux 5.7.19-2-MANJARO
Host: HP Probook 640 G1
CPU: Intel i7-4712MQ

**Installation Details**
Built from source using the Arch Package Build system. Details at https://aur.archlinux.org/packages/openvsp/

jrparks9469, 2020-09-15:
I decided to rebuild openvsp and redirect the output to a log file so you can see what my system is doing during the compilation. That file is attached. Hopefully it helps.
[log.txt](https://github.com/OpenVSP/OpenVSP/files/5228084/log.txt)

jrparks9469, 2020-09-15:
As an additional note, when I attempt to export a step file in Linux Mint, I do not receive the segmentation fault in the same way I do on an Arch-based distribution.

acxz, 2020-11-16:
@jrparks9469 I went ahead and updated the pkgbuild with the same build process used for debian based systems. I believe the issue is fixed but if you could test and comment on it that would be great.

justingravett, 2020-12-04:
I've been scratching my head over this issue all week, and have found that there is something strange with the Arch Linux package build system that is breaking STEPCode. OpenVSP compiles, but the segmentation fault still occurs any time VSP tries to call a function from STEPCode. 

I created a bash script to build OpenVSP and copied the contents of it into the build function of the PKGBUILD. Running the bash script and calling "makepkg" both completed successfully, but the segmentation fault still occurred in the package build. I attached the bash script, but had to change the file extension to upload to GitHub.

I'm really not sure how to fix the Arch Linux package build, but luckily this issue is specific to STEP exports with everything else seeming to work fine. Unless anyone has a suggestion, I suppose we can just mark this as a known bug with the workaround being to compile OpenVSP through the attached script instead of the package build if the STEP export feature is needed.

[build_vsp.txt](https://github.com/OpenVSP/OpenVSP/files/5644545/build_vsp.txt)

acxz, 2020-12-07:
hmm the Arch Linux package issue could be due to the fact that it is not built in Release mode. I'll add that in rq.

acxz, 2020-12-07:
@justingravett the Arch Linux package is now built in Release mode, can you check if you can still reproduce the issue? Thank you.

justingravett, 2020-12-07:
Just tried out the latest but am still getting the segmentation fault on STEP export. One other note is that I've been removing stepcode from the list of dependencies in the PKGBUILD file, since the STEPCode library is bundled with OpenVSP.

acxz, 2020-12-07:
> Just tried out the latest but am still getting the segmentation fault on STEP export.

Hmm welp I'm outta ideas.
Just to make sure, you did a clean build right?

> One other note is that I've been removing stepcode from the list of dependencies in the PKGBUILD file, since the STEPCode library is bundled with OpenVSP.

That should be fine, as even though I list stepcode as a dep, external stepcode doesn't get linked with OpenVSP as that is broken, see: https://github.com/OpenVSP/OpenVSP/pull/160

justingravett, 2020-12-07:
Correct, this was a clean build from a newly cloned directory of the latest package. Tested on a Manjaro 20.1 VM. I don't see anything wrong with the PKGBUILD file, so my only explanation is that the Arch Linux package manager is doing something weird with STEPCode that is causing it to break.

jrparks9469, 2020-12-17:
I'm seeing the same issue using the new PKGBUILD on the latest version of Arch (I haven't looked at it on Manjaro yet). I'm not entirely sure why using a PKGBUILD rather than running the commands manually would cause that large of a problem. I might try to look deeper into the problem if I get more time.

jrparks9469, 2020-12-18:
I've managed to get a working copy by: going into the srcdir after running makepkg, rebuilding the libraries and binaries manually, and running makepkg again. Some libraries and binaries appear to be rebuilt when I run makepkg the 2nd time, but I'm not 100% sure what is/isn't being rebuilt. When I run makepkg again, a new zip is created and the files in the pkgdir are replaced by working versions. I could rebuild both manually and using the PKGBUILD and create log files of each output to see what is being done differently because the problem seems to be related to build rather than obtaining the source or packaging the files.

jrparks9469, 2021-02-13:
I have have the build process saved to a plain text file for the manual build process and the arch build system. If I run a diff on those two files, the only apparent differences are Linking CXX executables readtest and mergetest:

AUR
```
/usr/bin/c++ -D_FORTIFY_SOURCE=2 -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -std=c++11 -Wall -O3 -DNDEBUG -Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now -rdynamic CMakeFiles/readtest.dir/tests/test_read.cpp.o -o readtest  libiges_static.a
```

Manual
```
/usr/bin/c++  -std=c++11 -Wall -O3 -DNDEBUG -rdynamic CMakeFiles/readtest.dir/tests/test_read.cpp.o -o readtest  libiges_static.a
```

ramcdona, 2023-03-12:
Closing this issue as it appears to be resolved.

closed 2023-03-12 by ramcdona