OpenVSP/OpenVSP #161: ImportError: Module use of python36.dll conflicts with this version of Python.

Ahorvath94, 2020-11-23:
Hi OpenVSP!

I think this is a very interesting project. I am trying to figure out how to get the Python API to work and I have looked in some of the example scripts. I am somewhat new at building C/C++ projects and am unfamiliar at how C/C++ wrappers work for Python. Anyways, I am currently using Python 3.7 for my project and would like to keep it this way. However, I get this message that says python36.dll conflicts with this version of Python, which leads me to believe my version of OpenVSP was built with Python 3.6 in mind. If I want to make the Python example scripts available for use with Python 3.7, how would I go about configuring or building OpenVSP for Python 3.7? Any help is appreciated! I am still trying to learn about OpenVSP but I think it looks very interesting and useful.

Thanks!

OS Windows 10
Pycharm community edition

Traceback (most recent call last):
  File "C:/Users/Owner/PycharmProjects/SMADLAB/OpenVSPTest.py", line 3, in <module>
    from openvsp.python.openvsp import openvsp as vsp
  File "C:\Users\Owner\AppData\Local\Programs\Python\Python37\lib\site-packages\openvsp\__init__.py", line 1, in <module>
    from .vsp import *
  File "C:\Users\Owner\AppData\Local\Programs\Python\Python37\lib\site-packages\openvsp\vsp.py", line 13, in <module>
    from openvsp import _vsp
ImportError: Module use of python36.dll conflicts with this version of Python.

ramcdona, 2020-11-23:
To do this, you'll have to build OpenVSP on your own machine -- and make sure to point CMake at the Python you are targeting.

There was a recent tutorial about building on Windows.  The [recording of the presentation is here](https://www.youtube.com/watch?v=UjGlyay-JMA&feature=youtu.be&t=5675).  There was also a follow-on presentation about building the API for different scripting languages including Matlab and Python, that [video starts here](https://youtu.be/UjGlyay-JMA?t=7443).

The main CMake variables you need to set are: PYTHON_EXECUTABLE, PYTHON_LIBRARY, PYTHON_INCLUDE_DIR, PYTHON_INCLUDE_PATH

In general, this kind of question is probably better addressed on the OpenVSP Google Group.

Ahorvath94, 2020-11-24:
Thanks a lot, both very helpful! Probably the best explanation of how to use Cmake I have ever seen in fact! I will post to Google groups.

closed 2020-11-24 by Ahorvath94