OpenVSP/OpenVSP #165: How to compute DegenGeom files from vsp file using the command line?

Alex031544, 2020-12-18:
Hello,

first of all: A big Thank You for sharing this outstanding tool!

I have a use case where I manipulate the vsp3 file in an automated way because the geometries are constantly changing.

If I understood it properly, however, I can only use vspAero with _DegenGeom files and not with the vsp3 file directly. In the tutorial it said, start openVSP and start vspAero and then the _DegenGeom files are created in the background and can then be used with `vspaero` from the command line.

What way can I use to create the _DegenGeom files from the command line without the GUI? (Python/C++ are fine)

Thanks in advance,
Alex

cibinjoseph, 2020-12-18:
You're right. The _DegenGeom files are required for VSPAero to work with.

In the OpenVSP installation folder, you'll find a directory named examples. 
The case, [examples/vspaero_ex/Swept_Wing_API](https://github.com/OpenVSP/OpenVSP/tree/main/examples/vspaero_ex/Swept_Wing_API) has a tutorial and an API script (Angelscript language) that describes how to save as a degenGeom file. The relevant section looks like this:

```C
 // Save as vsp3 and DegenGeom format
  WriteVSPFile("TR1208-API.vsp3", SET_ALL);
  SetComputationFileName(DEGEN_GEOM_CSV_TYPE, "TR1208-API_DegenGeom.csv");
  ComputeDegenGeom(SET_ALL, DEGEN_GEOM_CSV_TYPE);
```

Also, you might get a faster response posting queries like these on the [OpenVSP google group](https://groups.google.com/forum/#!forum/openvsp) too.

Alex031544, 2020-12-18:
Thank you for your quick response. I will give it a try.

Alex031544, 2020-12-18:
Btw: Running the example script throws an error:
```
OpenVSP/examples/vspaero_ex/Swept_Wing_API $ vsp -script TR1208.vspscript 
Error Code: 14, Desc: WriteResultsCSVFile::Invalid ID 
Error Code: 14, Desc: GetStringResults::Invalid ID 
  
 ALPHA  |  CL 
 ------ | ------ 
```

Alex031544, 2020-12-18:
I took your script and re-wrote this to
```
void main()
{
  // Alpha Sweep analysis using VSPAero
  ReadVSPFile("myproject.vsp3");

  // Save as vsp3 and DegenGeom format
  ComputeDegenGeom(SET_ALL, DEGEN_GEOM_CSV_TYPE);
  
  string myAnalysis = "VSPAEROSweep";
  SetAnalysisInputDefaults(myAnalysis);

  // List inputs, type, and current values
  PrintAnalysisInputs(myAnalysis);

  // Set inputs for VSPAero
  array<int> analysis_method(1, VORTEX_LATTICE);
  SetIntAnalysisInput(myAnalysis, "AnalysisMethod", analysis_method);
  
  array<double> Sref(1, 1.890);
  SetDoubleAnalysisInput(myAnalysis, "Sref", Sref);

  array<double> bref(1, 6.000);
  SetDoubleAnalysisInput(myAnalysis, "bref", bref);

  array<double> cref(1, 0.325);
  SetDoubleAnalysisInput(myAnalysis, "cref", cref);

  array<double> AlphaStart(1, -10.0);
  SetDoubleAnalysisInput(myAnalysis, "AlphaStart", AlphaStart);

  array<double> AlphaEnd(1, +10.0);
  SetDoubleAnalysisInput(myAnalysis, "AlphaEnd", AlphaEnd);

  int nAlpha = 11;
  array<int> AlphaNpts(1, nAlpha);
  SetIntAnalysisInput(myAnalysis, "AlphaNpts", AlphaNpts);

  // Run analysis and write results to CSV file
  string allResults = ExecAnalysis(myAnalysis);
  WriteResultsCSVFile(allResults, "Results.csv");

  // Print alpha and CL values
  array<string> sweepResults = GetStringResults(allResults, "ResultsVec");
  array<double> Alpha_res(nAlpha);
  array<double> CL_res(nAlpha);

  Print("");
  Print("ALPHA  |  CL");
  Print("------ | ------");
  for (uint8 i = 0; i < uint8(nAlpha); i++)
  {
    array<double> alpha_vec = GetDoubleResults(sweepResults[i], "Alpha");
    Alpha_res[i] = alpha_vec[int(alpha_vec.length()) - 1];

    array<double> cl_vec = GetDoubleResults(sweepResults[i], "CL");
    CL_res[i] = cl_vec[int(cl_vec.length()) - 1];

    Print(Alpha_res[i] + "      |   " + CL_res[i]);
  }

  // Check For API Errors
  while (GetNumTotalErrors() > 0)
  {
    ErrorObj err = PopLastError();
    Print(err.GetErrorString());
  }
}
```
and then I run it with:
```
vsp -script myproject.vspscript
		[input_name]        [type]	[#]	[current values-->]
		2DFEMFlag           0		1	0 
		ActuatorDiskFlag    0		1	0 
		AlphaEnd            1		1	10.000000 
		AlphaNpts           0		1	5 
		AlphaStart          1		1	-10.000000 
		AnalysisMethod      0		1	0 
		AutoTimeNumRevs     0		1	5 
		AutoTimeStepFlag    0		1	1 
		BatchModeFlag       0		1	1 
		BetaEnd             1		1	0.000000 
		BetaNpts            0		1	1 
		BetaStart           1		1	0.000000 
		CGGeomSet           0		1	6 
		Clmax               1		1	-1.000000 
		ClmaxToggle         0		1	0 
		FarDist             1		1	-1.000000 
		FarDistToggle       0		1	0 
		FixedWakeFlag       0		1	0 
		FromSteadyState     0		1	0 
		GeomSet             0		1	8 
		GroundEffect        1		1	-1.000000 
		GroundEffectToggle  0		1	0 
		HoverRamp           1		1	0.000000 
		HoverRampFlag       0		1	0 
		KTCorrection        0		1	1 
		MachEnd             1		1	0.000000 
		MachNpts            0		1	1 
		MachStart           1		1	0.044000 
		Machref             1		1	0.300000 
		ManualVrefFlag      0		1	0 
		MaxTurnAngle        1		1	-1.000000 
		MaxTurnToggle       0		1	0 
		NCPU                0		1	4 
		NoiseCalcFlag       0		1	0 
		NoiseCalcType       0		1	0 
		NoiseUnits          0		1	0 
		NumMassSlice        0		1	10 
		NumTimeSteps        0		1	120 
		NumWakeNodes        0		1	64 
		Precondition        0		1	0 
		ReCref              1		1	10000000.000000 
		RefFlag             0		1	0 
		Rho                 1		1	0.002377 
		RotateBladesFlag    0		1	0 
		Sref                1		1	1.890000 
		Symmetry            0		1	0 
		TimeStepSize        1		1	0.000278 
		UnsteadyType        0		1	0 
		Vinf                1		1	15.000000 
		Vref                1		1	15.000000 
		WakeNumIter         0		1	3 
		WingID              2		1	  
		Xcg                 1		1	0.085089 
		Ycg                 1		1	-0.000000 
		Zcg                 1		1	-0.059976 
		bref                1		1	6.000000 
		cref                1		1	0.325000 
Error Code: 14, Desc: WriteResultsCSVFile::Invalid ID 
Error Code: 14, Desc: GetStringResults::Invalid ID 
  
 ALPHA  |  CL 
 ------ | ------ 
An exception 'Index out of bounds' occurred 
```
It found, that the computation by vspAero is very fast. Using the GUI it took 5 minutes, but here no time goes by. It seems to me that no computation occurs.

cibinjoseph, 2020-12-19:
Seems like something that the script uses has changed in the recent versions. There are a few other examples available in the examples/scripts directory. I would suggest you look at them while I try to figure out what's going on.

Alex031544, 2020-12-20:
Same on a new build version of OpenVSP:
```
OpenVSP-3.23.0-Linux/vspaero_ex/Swept_Wing_API $ ../../vsp -script TR1208.vspscript 
Error Code: 14, Desc: WriteResultsCSVFile::Invalid ID 
Error Code: 14, Desc: GetStringResults::Invalid ID 
  
 ALPHA  |  CL 
 ------ | ------ 
An exception 'Index out of bounds' occurred 
```

Alex031544, 2020-12-20:
I wrote a small script to check if this angelscript ever run in the past:
```
#!/bin/bash

TAGS="OpenVSP_3.23.0"
TAGS="$TAGS OpenVSP_3.22.0"
TAGS="$TAGS OpenVSP_3.21.2"
TAGS="$TAGS OpenVSP_3.21.1"
TAGS="$TAGS OpenVSP_3.21.0"
TAGS="$TAGS OpenVSP_3.20.1"
TAGS="$TAGS OpenVSP_3.20.0"
TAGS="$TAGS OpenVSP_3.19.1"
TAGS="$TAGS OpenVSP_3.19.0"
TAGS="$TAGS OpenVSP_3.18.0"
TAGS="$TAGS OpenVSP_3.17.1"
TAGS="$TAGS OpenVSP_3.17.0"
TAGS="$TAGS OpenVSP_3.16.2"
TAGS="$TAGS OpenVSP_3.16.1"
TAGS="$TAGS OpenVSP_3.16.0"
TAGS="$TAGS OpenVSP_3.15.0"
TAGS="$TAGS OpenVSP_3.14.2"
TAGS="$TAGS OpenVSP_3.14.1"
TAGS="$TAGS OpenVSP_3.14.0"
TAGS="$TAGS OpenVSP_3.13.3"
TAGS="$TAGS OpenVSP_3.13.2"
TAGS="$TAGS OpenVSP_3.13.1"
TAGS="$TAGS OpenVSP_3.13.0"
TAGS="$TAGS OpenVSP_3.12.1"
TAGS="$TAGS OpenVSP_3.12.0"
TAGS="$TAGS OpenVSP_3.11.0"
TAGS="$TAGS OpenVSP_3.10.0"
TAGS="$TAGS OpenVSP_3.9.1"
TAGS="$TAGS OpenVSP_3.9.0"
TAGS="$TAGS OpenVSP_3.8.1"
TAGS="$TAGS OpenVSP_3.8.0"
TAGS="$TAGS OpenVSP_3.7.2"
TAGS="$TAGS OpenVSP_3.7.1"
TAGS="$TAGS OpenVSP_3.7.0"
TAGS="$TAGS OpenVSP_3.6.2"
TAGS="$TAGS OpenVSP_3.6.1"
TAGS="$TAGS OpenVSP_3.6.0"
TAGS="$TAGS OpenVSP_3.5.2"
TAGS="$TAGS OpenVSP_3.5.1"
TAGS="$TAGS OpenVSP_3.5.0"
TAGS="$TAGS OpenVSP_3.4.1"
TAGS="$TAGS OpenVSP_3.4.0"
TAGS="$TAGS OpenVSP_3.3.2"
TAGS="$TAGS OpenVSP_3.3.1"
TAGS="$TAGS OpenVSP_3.3.0"
TAGS="$TAGS OpenVSP_3.2.3"
TAGS="$TAGS OpenVSP_3.2.2"
TAGS="$TAGS OpenVSP_3.2.1"
TAGS="$TAGS OpenVSP_3.2.0"
TAGS="$TAGS OpenVSP_3.1.2"
TAGS="$TAGS OpenVSP_3.1.1"
TAGS="$TAGS OpenVSP_3.1.0"
TAGS="$TAGS OpenVSP_3.0.4"
TAGS="$TAGS OpenVSP_3.0.3"
TAGS="$TAGS OpenVSP_3.0.2"
TAGS="$TAGS OpenVSP_3.0.1"
TAGS="$TAGS OpenVSP_3.0.0"
TAGS="$TAGS OpenVSP_2.9.8"
TAGS="$TAGS OpenVSP_2.9.7"
TAGS="$TAGS OpenVSP_2.9.6"
TAGS="$TAGS OpenVSP_2.9.5"
TAGS="$TAGS OpenVSP_2.9.4"
TAGS="$TAGS OpenVSP_2.9.3"
TAGS="$TAGS OpenVSP_2.9.2"
TAGS="$TAGS OpenVSP_2.9.1"
TAGS="$TAGS OpenVSP_2.9.0"
[... 162 lines cut ...]
OpenVSP_3.22.0: test vspaero_ex/Swept_Wing_API ... done
#### OpenVSP_3.21.2 ####
OpenVSP_3.21.2: clone sources
OpenVSP_3.21.2: Prepare build files for the libraries in buildlibs
OpenVSP_3.21.2: build libs in buildlibs
OpenVSP_3.21.2: Set up and build OpenVSP
OpenVSP_3.21.2: build OpenVSP in build
OpenVSP_3.21.2: failed to build OpenVSP in build > abort
#### OpenVSP_3.21.1 ####
OpenVSP_3.21.1: clone sources
OpenVSP_3.21.1: Prepare build files for the libraries in buildlibs
OpenVSP_3.21.1: build libs in buildlibs
OpenVSP_3.21.1: Set up and build OpenVSP
OpenVSP_3.21.1: build OpenVSP in build
OpenVSP_3.21.1: Create OpenVSP package in build
OpenVSP_3.21.1: OpenVSP sucessfully build
OpenVSP_3.21.1: unpack OpenVSP
OpenVSP_3.21.1: test vspaero_ex/Swept_Wing_API
OpenVSP_3.21.1: execute ../../vsp -script TR1208.vspscript
OpenVSP_3.21.1: return value 127
OpenVSP_3.21.1: test vspaero_ex/Swept_Wing_API ... done
#### OpenVSP_3.21.0 ####
...
```
So. This script was introduced in OpenVSP_3.22.0, but no version was sucessful :(

Alex031544, 2020-12-20:
I started debugging it a bit and put some debug outputs in the code.

Here what I found:
```
#### OpenVSP_3.23.0 ####
OpenVSP_3.23.0: Prepare build files for the libraries in buildlibs
OpenVSP_3.23.0: build libs in buildlibs
OpenVSP_3.23.0: Set up and build OpenVSP
OpenVSP_3.23.0: build OpenVSP in build
OpenVSP_3.23.0: Create OpenVSP package in build
OpenVSP_3.23.0: OpenVSP sucessfully build
OpenVSP_3.23.0: unpack OpenVSP
OpenVSP_3.23.0: test vspaero_ex/Swept_Wing_API
OpenVSP_3.23.0: execute ../../vsp -script TR1208.vspscript
 myAnalysis: VSPAEROSweep
/home/alex/git/OpenVSP/OpenVSP/repo/src/geom_core/AnalysisMgr.cpp:ExecAnalysis:125: analysis: VSPAEROSweep
/home/alex/git/OpenVSP/OpenVSP/repo/src/geom_core/AnalysisMgr.cpp:ExecAnalysis:138: analysis_ptr: 0x55706c2a8480
/home/alex/git/OpenVSP/OpenVSP/repo/src/geom_core/AnalysisMgr.cpp:ExecAnalysis:140: analysis_ptr->Execute()
/home/alex/git/OpenVSP/OpenVSP/repo/src/geom_core/AnalysisMgr.cpp:Execute:1891:
/home/alex/git/OpenVSP/OpenVSP/repo/src/geom_core/AnalysisMgr.cpp:Execute:1894: veh: 0x55706c120090
/home/alex/git/OpenVSP/OpenVSP/repo/src/geom_core/AnalysisMgr.cpp:Execute:1905: geomSetOrig: 1
/home/alex/git/OpenVSP/OpenVSP/repo/src/geom_core/AnalysisMgr.cpp:Execute:1911: nvd: 0x5570723aff40
/home/alex/git/OpenVSP/OpenVSP/repo/src/geom_core/AnalysisMgr.cpp:Execute:2024: VSPAEROMgr.m_AlphaStart: 0
/home/alex/git/OpenVSP/OpenVSP/repo/src/geom_core/AnalysisMgr.cpp:Execute:2031: VSPAEROMgr.m_AlphaEnd: 9
/home/alex/git/OpenVSP/OpenVSP/repo/src/geom_core/AnalysisMgr.cpp:Execute:2092: VSPAEROMgr.m_NCPU: 0
/home/alex/git/OpenVSP/OpenVSP/repo/src/geom_core/AnalysisMgr.cpp:Execute:2298: VSPAEROMgr.ComputeSolver
/home/alex/git/OpenVSP/OpenVSP/repo/src/geom_core/VSPAEROMgr.cpp:ComputeSolver:1670
/home/alex/git/OpenVSP/OpenVSP/repo/src/geom_core/VSPAEROMgr.cpp:ComputeSolver:1676: m_DegenGeomVec.size() == 0
/home/alex/git/OpenVSP/OpenVSP/repo/src/geom_core/AnalysisMgr.cpp:Execute:2300: VSPAEROMgr.ComputeSolver resId:
/home/alex/git/OpenVSP/OpenVSP/repo/src/geom_core/AnalysisMgr.cpp:ExecAnalysis:142: res
 allResults:
Error Code: 14, Desc: WriteResultsCSVFile::Invalid ID
Error Code: 14, Desc: GetStringResults::Invalid ID

 ALPHA  |  CL
 ------ | ------
An exception 'Index out of bounds' occurred
OpenVSP_3.23.0: return value 0
OpenVSP_3.23.0: test vspaero_ex/Swept_Wing_API ... done
```

It seems, that `VSPAEROMgr.cpp:ComputeSolver:1676: m_DegenGeomVec.size() == 0` is not what it should be.

Here the corresponding code snipped:
```
string VSPAEROMgrSingleton::ComputeSolver( FILE * logFile )
{
    std::cout << __FILE__ << ":" << __FUNCTION__ << ":" << __LINE__ << std::endl; // TODO remove
    Update(); // Force update to ensure correct number of unstead groups, actuator disks, etc when run though the API.
    UpdateFilenames();

    if ( m_DegenGeomVec.size() == 0 )
    {
        std::cout << __FILE__ << ":" << __FUNCTION__ << ":" << __LINE__ << ": m_DegenGeomVec.size() == 0" << std::endl; // TODO remove
        // Note: Using errMsgData.m_String = "Error" will fail when this function
        // is not called from the main VSP thread
        MessageData errMsgData;
        errMsgData.m_String = "VSPAEROSolverMessage";
        errMsgData.m_StringVec.emplace_back( string( "Error: No Geometry in VSPAERO Set\n" ) );
        MessageMgr::getInstance().SendAll( errMsgData );

        return string();
    }

    if ( m_CpSliceFlag() )
    {
        ClearCpSliceResults();
    }
    if ( m_BatchModeFlag.Get() )
    {
        std::cout << __FILE__ << ":" << __FUNCTION__ << ":" << __LINE__ << ": ComputeSolverBatch" << std::endl; // TODO remove
        return ComputeSolverBatch( logFile );
    }
    else
    {
        std::cout << __FILE__ << ":" << __FUNCTION__ << ":" << __LINE__ << ": ComputeSolverSingle" << std::endl; // TODO remove
        return ComputeSolverSingle( logFile );
    }

    return string();
}
```

Alex031544, 2020-12-20:
Soooo. I add `VSPAEROMgr.ComputeGeometry();` to `string VSPAEROSweepAnalysis::Execute()` right before `VSPAEROMgr.ComputeSolver`:
```
        //==== Execute Analysis ====//
        VSPAEROMgr.ComputeGeometry();
        resId = VSPAEROMgr.ComputeSolver(stdout);
```
This seems to work... :)

Can you double-check this?

Alex031544, 2020-12-20:
I add a related pull request https://github.com/OpenVSP/OpenVSP/pull/166

cibinjoseph, 2020-12-21:
Hey good job on tracking down the bug. I was busy a bit. Seems like ComputeGeom has to be explicitly called before the VSPAero solver; which is odd though, since the DegenGeom file is already being generated.

However, instead of calling `VSPAEROMgr.ComputeGeometry()` directly, I think it might be appropriate to invoke it using the API as an analysis like so:
```C
ReadVSPFile("TR1208-API.vsp3");

// Run CompGeom to generate geometry
string compGeom = "VSPAEROComputeGeometry";
SetAnalysisInputDefaults(compGeom);
string compGeom_results = ExecAnalysis(compGeom);

// Alpha sweep analysis using VSPAero
```

Alex031544, 2020-12-21:
I add a related pull request #167 and close #166.

cibinjoseph, 2020-12-21:
@Alex031544 You'll need to also include a call to `ClearVSPModel()` to make it work. Check PR #168

closed 2020-12-27 by Alex031544