OpenVSP/OpenVSP #171: Compile Error in apitest_main.cpp

jstults, 2021-01-03:
I cloned the git source into a directory for a build.  I've followed [1] to get the dependencies built or configured successfully with this cmake, and then make.  

$ cmake -DVSP_USE_SYSTEM_LIBXML2=true -DVSP_USE_SYSTEM_FLTK=true -DVSP_USE_SYSTEM_GLM=true -DVSP_USE_SYSTEM_GLEW=true -DVSP_USE_SYSTEM_CMINPACK=true -DVSP_USE_SYSTEM_LIBIGES=false -DVSP_USE_SYSTEM_EIGEN=true -DVSP_USE_SYSTEM_CODEELI=false -DVSP_USE_SYSTEM_CPPTEST=true ../OpenVSP/Libraries -DCMAKE_BUILD_TYPE=Release

Then this configuration for the main build without errors:  

$ cmake ../OpenVSP/src/ -DVSP_LIBRARY_PATH=/home/josh/OpenVSP_git/buildlibs -DCMAKE_BUILD_TYPE=Release 

But I get a compile error trying to make:    

In function ‘bool run_tests()’:  OpenVSP/src/vsp/apitest_main.cpp:34:17: error: cannot convert ‘std::unique_ptr<Test::Suite>’ to ‘std::auto_ptr<Test::Suite>’ 

Any tips on fixing this appreciated; thanks!   

[1] http://openvsp.org/wiki/doku.php?id=ubuntu_instructions

ramcdona, 2021-01-05:
It is probably better to discuss this sort of thing on the Google Group -- then we can take the final resolution to GitHub.

As a workaround, I would suggest you try running make with a -k.  This will keep going, despite errors.  If the only error is in the apitest, then this should get you to a vsp executable so you can keep making progress.

gcc 10.2.1 is very new, I don't think I have tried to build with such a new version yet.

jstults, 2021-01-05:
Got a working executable with make -k.  Just joined the google group. 

Thanks.

jstults, 2021-01-05:
Rob,

What is the latest version of gcc that works successfully for you?

V/R,
Josh

closed 2024-05-11 by ramcdona