OpenVSP/OpenVSP #172: Python import error - 0.23.0, OS X Big Sur

oli-hall, 2021-01-04:
Hi, I'm coming across an import error in 0.23.0 with any `openvsp` Python import. I was previously running 0.21.2 with zero issues, but decided to upgrade to the latest version after coming across #162. 

The import error manifests as a `Symbol not found _PyCMethod_New`:

```
File "<project>/design.py", line 6, in <module>
    import openvsp as vsp
File "<project>/bin/env/lib/python3.7/site-packages/openvsp/__init__.py", line 1, in <module>
    from .vsp import *
File "<project>/bin/env/lib/python3.7/site-packages/openvsp/vsp.py", line 13, in <module>
    from . import _vsp
ImportError: dlopen(<project>/bin/env/lib/python3.7/site-packages/openvsp/_vsp.so, 2): Symbol not found: _PyCMethod_New
  Referenced from: <project>/bin/env/lib/python3.7/site-packages/openvsp/_vsp.so
```

As a workaround, I've reverted back to the 0.21.2 `openvsp` wheel, and that's working just fine.

ramcdona, 2021-01-05:
In addition to other fixes, the latest version required that we migrate our build system from TravisCI to GitHub Actions.  This means some of the details of the OS, compiler, libraries, and Python that we build against may have changed.

We build with the [MacOS 10.15](https://github.com/actions/virtual-environments/blob/main/images/macos/macos-10.15-Readme.md) image.

I would try matching that Python (3.9.1) as my first try.

Also, this kind of question is typically better resolved on the Google Group.  We can then pull the final solution to GitHub.

oli-hall, 2021-01-05:
Thanks for the reply, and apologies if I raised this in the wrong place! I'm using `pyenv`, which hasn't yet finalised its Python 3.9.1 release, but was able to test with 3.9.0, and this issue is entirely resolved (for others who may come across this, I was running Python 3.7.5 when I encountered the above issue).

I'll close this off now, as it seems I was mistaken about this being a problem!

closed 2021-01-05 by oli-hall