OpenVSP/OpenVSP #178: Subsurface Control Surfaces cover wrong spanwise locations

GEbb4, 2021-03-18:
When setting the spanwise U location of the start and end of a control surface, the fraction starts and ends at the wing camber line, rather than the span - so the number of panels down the end of the aerofoil section (for a square wing) affects where the U location 0.9 is, for example.

**Steps to reproduce:**
1. Open OpenVSP 3.23.0
2. Create a new wing which is straight, unswept, untapered, with constant chord 1m and a total span of 2m (so a single section with span 1m, with XZ reflection on).
3. Reduce the number of U interpolated XSecs to 2, which is one at the root and one at the tip (not strictly necessary but it makes it easier to see).
4. Add a control surface subsection, then set the Start U to 0.1 and the End U to 0.8. The control surface will cover the entire trailing edge because the vertical distance from the camber line to the wing surface is wrongly included in the U panelling.

justingravett, 2021-03-22:
I don't believe this is bug. The U parameter for control surfaces is based on the number of sections that the wing has. By default a wing has 3: a flat root end cap, a main wing section, and a flat tip end cap. Each section is provided an equal portion of the U parameter for the entire surface. Therefore, U will start at 1/3 and end at 2/3 for the main section of the default wing. Adding sections or changing the end cap type to "None" will change the start and end U parameter value for a particular section.

GEbb4, 2021-03-23:
For a control surface, what's the benefit or practical meaning to being able to have it start/end halfway down the end cap? This seems like inconsistent behaviour. If there is a meaning to having a control surface only take up part of the depth of the wing, then it can only be done for the entire span, otherwise it can only be part of the top surface?

But in any case, I think it would be far more helpful for the input box to accept (or have an option for) "spanwise position" directly, which wouldn't include the end caps. Most aircraft control surfaces I've seen are quoted as e.g. "50% to 90% span". Whilst the conversion isn't _too_ difficult, for multi-section wings it's an extra layer of obfuscation which is unnecessary, counter-intuitive, and likely quite error-prone.

As an aside, it's also quite confusing for the current section to be labelled as "Spanwise", but then not actually accept spanwise locations...

ramcdona, 2021-03-23:
I agree that an additional input type for control surfaces that does the math to give you a true %span input would be nice to have.

This idea was already applied for the chordwise input for control surfaces -- that took quite a bit of work to do.  Fixing up the spanwise one would probably be a bit easier actually.

OpenVSP exposes the U,V coordinates to users a number of other places -- U,V attachment, other subsurface types, point probe and line measure, and CFDMesh sourcing to name a few.  So, this input is consistent with all the other uses throughout OpenVSP.

While the word spanwise does not strictly mean %span -- it is really to differentiate that it is in the spanwise direction, not the chordwise direction, or some other direction.

The U parameter runs along each section of panel.  For a single-segment wing, there are likely three sections -- the root cap, the wing itself, and the tip cap.  If you turn the root and tip cap off, you'll get the behavior you're looking for.

Each section has an equal range of U.  So, a single segment wing with root/end caps (3 surface sections), each section will span 1/3 of U.  So, the main wing itself will run from 0.3333 to 0.6666.  If you want to have a % span coordinate *call it eta) for a single-segment wing, then you should use the equation U=(1+eta)/3.  For an n-segment wing, this generalizes to U=(1+eta*n)/(n+2).

These kinds of questions are great to ask on the OpenVSP Google Group.

Hope this helps.

TheSoundMan000, 2021-04-05:
I've just come here to flag this exact issue thinking as @GEbb4 did that something buggy was going on. Thanks @justingravett for explaining how this works, is this documented somewhere or just something known by everyone?

I can see the utility for the functionality provided but its a bit confusing. Until this particular issue log I was fairly convinced it was broken, its quite counter intuitive. Perhaps an alternative implementation could take the wing section and the percentage through each section to make things a little clearer? With the current implementation if you add/remove a section you have to redo all the math. This would isolate that issue and make it (atleast I think) somewhat more intuitive.

ramcdona, 2021-04-05:
Our best documentation is the videos in the OpenVSP Ground School.  I am sure it is mentioned in there.

Yes, if you add / remove wing sections, your control surfaces will jump around.  There are quite a few places in OpenVSP where topological model changes will have perhaps unexpected effects.

Usually this isn't a problem -- most parametric design studies work via the parameters on a fixed topology.  So, the easiest way to go is to build up your model, get all the topological things done, then come back through and add subsurfaces, U,W based attachment, links, advanced links, and anything else that is sensitive to topology changes.

There are some possible ways to make the control surface behavior more familiar, but we also have to respect compatibility with files that are already out there.

closed 2021-06-11 by ramcdona