OpenVSP/OpenVSP #182: IBL3?

RibeiroAndre, 2021-05-11:
Hi,
I read on a forum that back in 2016 or so it was mentioned that Drela's IBL3 would be implemented into VSPAero.
I'm curious about how that went. Is there still a plan to implement it?
Thanks!

ramcdona, 2021-05-28:
I don't think that was ever in the plan.  Can you post a link to the conversation?  Perhaps seeing it in context will jog my memory.

Most OpenVSP discussion happens in the Google Group, you might post future questions there.

RibeiroAndre, 2021-05-28:
https://www.rcgroups.com/forums/showthread.php?2095696-XoptFoil-airfoil-optimization-with-XFoil/page23
" IBL3 actually came up in a talk titled "VSPAERO Verification & Next Steps" I just watched a few weeks ago. The talk was given at NASA's August 2016 OpenVSP Workshop, a conference for their relatively new, open-source conceptual aircraft design program.

According to Dave Kinney, NASA Ames is "overseeing some work" on IBL3 and considering including it with the panel code solver sometime during the next year. "
The link to the video doesn't work anymore.

ramcdona, 2021-05-28:
Thanks for the link.

I almost recall that conversation.  The video should work, but it requires Flash to play back, so it probably fails with any modern browser.

I'll try to ask Dave if anything came of that -- but to my knowledge, there has been no such progress.

closed 2021-05-28 by ramcdona