OpenVSP/OpenVSP #185: Installation on Ubuntu 20.04 fails all the time.

pmsagdeo, 2021-06-06:
Installation fails with the message shown in screenshot attached. 
[OpenVspInstallationError06062021.odt](https://github.com/OpenVSP/OpenVSP/files/6604912/OpenVspInstallationError06062021.odt)

Has happened all the time. I have not been able to install it on a Linux machine with Ubuntu 20.04.  Please advise.  Thank you.

cibinjoseph, 2021-06-07:
Looks like an issue during post-installation.
Can you try using `apt` in the terminal to install the deb package like so:
`sudo apt install ./OpenVSP-3.24.0-Ubuntu-20.04_amd64-Python3.6.deb`

pmsagdeo, 2021-06-07:
OK. Will give it a shot and let you know.  Thanks. 

 

Pradip 

 

This email has been checked for viruses by AVG.
https://www.avg.com

pmsagdeo, 2021-06-08:
Here's the result:

***@***.***:~/Downloads$ sudo apt install 
./OpenVSP-3.24.0-Ubuntu-20.04_amd64-Python3.6.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'openvsp' instead of 
'./OpenVSP-3.24.0-Ubuntu-20.04_amd64-Python3.6.deb'
openvsp is already the newest version (3.24.0).
The following packages were automatically installed and are no longer 
required:
   adobe-flash-properties-gtk carla-lv2 carla-vst distro-info gxplugins-data
   lib32gcc1 libclucene-contribs1v5 libclucene-core1v5 libcmis-0.5-5v5 
libeot0
   libexttextcat-2.0-0 libexttextcat-data libfprint-2-tod1 libgfortran4
   libhypre-2.13.0 libjuh-java libjurt-java liblangtag-common liblangtag1
   libllvm10 libllvm10:i386 libllvm9 libllvm9:i386 libmhash2 libmumps-5.1.2
   libmythes-1.2-0 liborcus-0.15-0 libptscotch-6.0 librasqal3 librdf0
   libridl-java libscalapack-openmpi2.0 libscotch-6.0 libsuperlu-dist5
   libuno-cppu3 libuno-cppuhelpergcc3-3 libuno-purpenvhelpergcc3-3 
libuno-sal3
   libuno-salhelpergcc3-3 libunoloader-java libxmlsec1 libxmlsec1-nss
   openclipart-png uno-libs-private ure
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 34 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Setting up openvsp (3.24.0) ...
Setting up symlinks ...
ln: failed to create symbolic link '/usr/local/bin/vsp': No such file or 
directo
ry
dpkg: error processing package openvsp (--configure):
  installed openvsp package post-installation script subprocess returned 
error ex
it status 1
Errors were encountered while processing:
  openvsp
E: Sub-process /usr/bin/dpkg returned an error code (1)

What should I do now?  Thanks.

Pradip

cibinjoseph, 2021-06-08:
It appears that the directory `/usr/local/bin` is not available on your system. This may have occurred when an other package was removed and its cleanup script mistakenly removed the directory. 
You can verify this with the command:
`ls -ld /usr/local/bin`

It should return something like:
`ls: cannot access '/usr/local/bin': No such file or directory`

If this is indeed the case, create that directory with:
`mkdir -p /usr/local/bin`

and then re-install via the software center or using the apt command 
`sudo apt install ./OpenVSP-3.24.0-Ubuntu-20.04_amd64-Python3.6.deb`

ramcdona, 2021-08-10:
Should be fixed by 5a563ca2a7d714d

closed 2021-08-10 by ramcdona