OpenVSP/OpenVSP #189: Userparameter with advanced link does not update the geometry properly on the first try

mcriedel, 2021-06-16:
When updating some user-defined parameters, which are used within the advanced linker, the first update gets not completely processed. Only after changing the value for a second time or forcing an update externaly, the geometry gets updated correctly.

Example: The attached file contains a joint wing. When opening the user parms window and adjust the variable AR_ratio to another value (like  = 5), it seems like nothing has happened. After setting the parameter to another value ( = 4.9), the rear wing gets updated. The update can also be forced by selecting the rear wing in the object tree. 
The same happens with the python API, but here, the behavior can be better observed: When the parameter "AR_ratio" is set, the advanced link "rw_AR = AR_ratio*fw_AR;" gets triggered and rw_AR is set properly, which is an alias for parameter "Aspect" in the group "XSec_1" of the container "wingRear". But the overall parameter "TotalAR" in the group "WingGeom" of the same container is not updated on the first try. Also a vsp.Update() does not change it. Only after setting the value for a second time (and with a different value because otherwise openVSP recognizes, that the value has not changed), the update process works correctly.

Version: This bug can be reproduced by using OpenVSP 3.24.0 (GUI) and OpenVSP 3.23.0 (Python API) [other versions were not tested]

This may look like a minor bug when using the GUI but it is very tedious when using the API. We use the API for automating parameter studies of joint wings for a PhD project and would appreciate it, if this issue could be fixed.

[OpenVSP_Bug_Uservars.vsp3.txt](https://github.com/OpenVSP/OpenVSP/files/6662882/OpenVSP_Bug_Uservars.vsp3.txt)

ramcdona, 2021-06-18:
Thanks for bringing this to our attention.  This is a great place to make sure a bug does not get lost, but it also doesn't get checked that regularly.  So, you may also want to bring issues like this to the Google Group.

justingravett, 2021-06-22:
This seems to have actually been fixed when addressing some of the other recent issues reported. I can replicate your issue on 3.24.0 but everything seems to be working correctly on my branch with the fixes. I can push this branch to my public repo or email you the compiled binaries if you'd like to test it out and confirm it solves the problems you've reported.

mcriedel, 2021-06-29:
Thank you for your answers. We built a binary based on your commit c089bc6 and for me it does not process the change correctly. Now, instead of remaining unchanged, the "TotalAR" parameter decreases slightly (It should increase). It still needs a second try to get updated correctly.

justingravett, 2021-06-30:
Did you test my branch through the GUI or just the API? Maybe I'm missing something, so here are the steps I'm following to replicate the issue in the GUI: 

1. Open OpenVSP_Bug_Uservars.vsp3
2. Observe Total Aspect Ratio of "wingRear" is 7.599 at this time (Plan tab)
3. Change the User Parm "AR_ratio" from 1.0 to 5.0. 
4. Observe AR of "wingRear" section 1 change from 5.019 to 25.096. Observe Total Aspect Ratio of "wingRear" change to 37.914.

Is this not the behavior I should be expecting? Am I missing a parameter that should be changing? 

If you are only able to replicate this through the API please let me know and I'll come up with a script to test it. If you are however able to provide your script I would greatly appreciate it. 

One other note is that you'll need to make sure you reinstall the python API packaged with with the unreleased branch I pointed you to to get the fix included when VSP in imported in Python. Otherwise a *.vspscript can be used as a workaround to the Python API for generating a test case.

mcriedel, 2021-06-30:
I tested both, the API and the GUI. You can reproduce the bug if you skip step 2). When interacting with the geometry tree, somehow the update is forced (If you do not, it stays the same untill it is changed again). You can clearly see this update forcing when updating the variable and observe the geometry while selecting the wing in the tree.

justingravett, 2021-06-30:
Even with skipping step 2 I'm still not seeing the behavior you are describing. When I update the "AR_ratio" User Parm, "wingFront" does not change but "wingRear" does in the main screen. If I have the "wingRear" selected I can see Total Aspect Ratio and the AR of section 1 updating immediate when the User Parm is changed. 

In the released 3.14.0, the geometry does not change when the "AR_ratio" User Parm is changed until the "wingRear" is selected as you've described. I apologize if I'm misunderstanding something, but it still seems like my branch fixes this bug. 

When you built my unreleased version of the code did you just cherry pick the last commit c089bc6 or checkout the entire branch? Thanks for your help explaining all of this.

mcriedel, 2021-06-30:
We used a checkout for the branch but the latest commit is c089bc6. 
To be sure that we are on the same page; what I do is:
1) open OpenVSP
2) Go to File -> open -> select the provided file
3) Go to Model -> User Parms -> Adjust -> AR_ratio
4) Set the value to 5 by using the text field and pressing enter -> The geometry is not updated
5) Set the value to 4.95 and press enter -> The geometry gets updated. 
Alternative to Step 5): Select wingRear in the Geom Browser.

The fix for #190 works well for the GUI (the API still crashes), so I am quite sure that we compiled the right version.

justingravett, 2021-06-30:
It looks like I've misunderstood, but this is now just an issue with the API and the GUI update issue has been fixed. Is that correct? On my system (Windows x64), the geometry does in fact update in the GUI in Step #4 above.  

I'll write a script to test this from the API. Since I won't be able to see if the geometry is updating I'll approach this by setting the "AR_ratio" User Parm, calling "Update()", and then checking the Parm Value of the total AR of the rearWing. That should be equivalent to replicating the issue in the GUI.

mcriedel, 2021-06-30:
No, at least for me, step 4) does not update the GUI (also Win x64). But we will rebuild the binaries and test it on another PC to be sure that this is not a build error and that we are using the latest commit. Thank you for your efforts!

RY3A1R, 2021-07-01:
Hi guys, 

thanks for looking into this!
mcriedel and me just tried to replicate the issue with a new build - still the same results.
We checked out JGravett/3.25.0_GeneralImprovements, the latest commit being c08bc6.

I would like to rephrase mcriedels reproduction guide and add one more observation:

1. open OpenVSP
2. Go to File -> open -> select the provided file
3. Go to Model -> User Parms -> Adjust -> AR_ratio
4. Set the value to 5 by using the text field, do not touch the slider or select anything
    and pressing enter -> The geometry updates somewhat
    The wing tip moves forward slightly
5. Set the value to 5.000001 and press enter -> Rear Wing Aspect Ratio changes, Tip moves aft
6. Set the value to 5 and press enter -> Tip moves to correct location
7. Set the value to 5.000001 and press enter -> nothing changes visibly (as we would expect)
8. The fore-aft wing tip movement cannot be fixed by selecting the rear Wing in the geom browser.

The wing tip fore-aft movement issue seems to be unrelated to the Aspect Ratio change issue, as it does not only occur after file load. Each later change in AR_ratio requires 2 Updates of the parameter to work correctly. (Updates only occuring when the paramter actually changes, so something like 1.0 -> 1.000001 is required.
My first guess was some sort of iteration going on inside the Adv Link, with out link code relying on one of its outputs to work correctly. However, I can't find any issues there.

Can the Aspect Ratio setting or the fore-aft movement issue stem from the fact that we set a parameter directly that is not a fundamental parameter of the wing, but one that is calculated based on the basic wing parameters?

Cheers,
Julius

mcriedel, 2021-07-01:
> The fix for #190 works well for the GUI (the API still crashes), so I am quite sure that we compiled the right version.

This crash turned out to be due to a wrong API setup. After reinstalling the API, #190 seems to be fixed for the API, too. Sorry for the confusion. Since I am not familiar with your bug tracking policy, I will leave it to you to close the issues.

justingravett, 2021-07-08:
Glad to hear and no worries for any confusion! I'll plan on leaving this open for now but can manually close it once the next release gets out.

justingravett, 2021-08-23:
I believe this issue has been closed with the release of 3.25.0 so I'll go ahead and close it manually unless I hear from anyone that the problem still persists.

closed 2021-09-07 by justingravett