OpenVSP/OpenVSP #193: Adding Conformal Volume to blended wing with custom aerofoil crashes the software

harishrajan96, 2021-07-06:
If a conformal volume is added to a blended wing body with custom airfoil profile (in my case eppler339), openvsp crashes. However if the conformal body is added with default symmetric aerofoil profile followed by modifying the aerofoil then the software does not crash. However adding more conformal body once the profile is modified (for secondary tanks) results in vsp crashing.

justingravett, 2021-07-08:
Would you be able provide your *.vsp3 file? I just tried replicating the issue with a blended wing test case using only b737d airfoil files (included with the distribution). I tried adding dihedral blend as well but was still able to add a conformal component. There may be an issue with your particular geometry that is throwing off the algorithm. I'm happy to look into it further if you can provide any additional information. [
[image]
](url)

harishrajan96, 2021-07-18:
Hello,

Sorry for the late reply, I redid the whole aircraft and now it is able to generate the conformal geometry. I want to use only the conformal geometry in fusion360, exporting from vsp gives a file which contains only bodies and not surfaces which I can stitch to create a solid in fusion 360.

closed 2023-03-12 by ramcdona