OpenVSP/OpenVSP #194: STP file export does not have surfaces information

harishrajan96, 2021-07-18:
Hello,

I am trying to import one of my OpenVSP models (winggeom and conformal) into FreeCAD. In order to do this I export my model from OpenVSP as a .stp file and after that I import that into FreeCAD. But I find that the model does not have any surfaces information, in other words I am not able to modify the surfaces in FreeCAD. Is there any solution or work around for it ?

ramcdona, 2021-07-20:
Does the geometry import at all?  What information are you expecting that is not there?

If you are expecting the geometry to still be parametric -- defined in terms of things like sweep angle, aspect ratio, and span -- that is unfortunately impossible.

When a model is exported from OpenVSP, all of the parametric information used to construct that model is lost.  The STEP and IGES standards can represent Bezier surfaces via control points, but not any information about how those control points were placed.

I am not familiar with FreeCAD, but you may need to convert the surfaces into a solid in order to do some kinds of editing and modifications to the geometry.

In addition to pure surface exports, OpenVSP can also export a geometry as a solid -- to do this, go to the Analysis menu, Trimmed Surfaces...  Make sure the BREP Solid option is on if you want to output a solid.  Otherwise, it will write out disconnected surfaces.

ramcdona, 2021-07-20:
One more thing...  This kind of question is generally best asked on the OpenVSP Google Group.  That way others can find your question and learn from the answer.

closed 2023-03-12 by ramcdona