OpenVSP/OpenVSP #196: Build Failure Due to -Werror=format-security

goznorlives, 2021-08-29:
Hate to bug you all with something this trivial, but I'm running into an issue on Manjaro Linux that I'm not quite sure how to fix on my own. I'd love to be able to submit a pull request, but I haven't done C++ in nearly 20 years...

I successfully installed an AUR package for OpenVSP a while back, but was unable to update it at a certain point. I've been looking at it for the last couple of days and it would seem that the issue is being caused by my system having -Werror=format-security set by default. This is causing the following compile error:

<pre>[ 80%] <font color="#00AA00">Building CXX object vsp_aero/solver/CMakeFiles/vspaero.dir/ComponentGroup.C.o</font>
In file included from <b>/var/tmp/pamac-build-ericf/openvsp/src/OpenVSP-OpenVSP_3.25.0/src/vsp_aero/solver/utils.H:17</b>,
                 from <b>/var/tmp/pamac-build-ericf/openvsp/src/OpenVSP-OpenVSP_3.25.0/src/vsp_aero/solver/ComponentGroup.H:15</b>,
                 from <b>/var/tmp/pamac-build-ericf/openvsp/src/OpenVSP-OpenVSP_3.25.0/src/vsp_aero/solver/ComponentGroup.C:7</b>:
/var/tmp/pamac-build-ericf/openvsp/src/OpenVSP-OpenVSP_3.25.0/src/vsp_aero/solver/VSPAERO_TYPES.H: In instantiation of ‘<b>void PRINTF(const char*, const Args&amp; ...) [with Args = {}]</b>’:
<b>/var/tmp/pamac-build-ericf/openvsp/src/OpenVSP-OpenVSP_3.25.0/src/vsp_aero/solver/ComponentGroup.C:461:15:</b>   required from here
<b>/var/tmp/pamac-build-ericf/openvsp/src/OpenVSP-OpenVSP_3.25.0/src/vsp_aero/solver/VSPAERO_TYPES.H:76:9:</b> <font color="#FF5555"><b>error: </b></font>format not a string literal and no format arguments [<font color="#FF5555"><b>-Werror=format-security</b></font>]
   76 |   <font color="#FF5555"><b>printf(format, Argument(args) ...)</b></font>;
      |   <font color="#FF5555"><b>~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
</pre>

ramcdona, 2021-08-30:
You might try adding a -k to make.  This will cause it to keep running despite errors.

In this case, VSPAERO is throwing the error -- but you might still be able to get OpenVSP working with this limitation.

I'm not sure if that will be sufficient for your situation, but if all you need is OpenVSP, it might get you limping along.

Otherwise, I recommend we wait on searching for a solution.  VSPAERO is undergoing a transition to support built-in calculations of derivatives. This has required replacing all floating point numbers 'double' with a macro VSPDOUBLE that is defined differently when building the derivative version.  This means that all printf statements need to be able to handle different types depending on whether VSPDOUBLE is a double or something else.  This resulted in the code that caused the warnings you're running into (which are promoted to errors on your system).

I am not sure if this will be the long-term solution or if something more elegant will be worked out.  I do know that this is currently in a state of flux.

goznorlives, 2021-08-30:
Thank you for the explanation. I'll give that a try. I'm really only an interested hobbyist, so no pressure on requirements. Thank you for your reply.

closed 2021-08-30 by goznorlives

goznorlives, 2021-08-31:
Just for posterity, I seem to have fixed this by exporting CFLAGS=-Wno-format and CXXFLAGS=-Wno-format to the environment before running cmake. Thanks again...