OpenVSP/OpenVSP #197: VSPaero - blank component 0 in .lod file (OpenVSP 3.25.0)

WillJLH, 2021-08-31:
Hi,

I've recently upgraded from OpenVSP 3.23.0 to 3.25.0. I'm noticing that there's now a blank _"component 0"_ showing in my .lod files. See below for a direct comparison taken from the .lod files for the same model and VSPaero set-up in versions 3.23.0 and 3.25.0 respectively:

From 3.23.0:

> Comp      Component-Name                             Mach       AoA      Beta       CL        CDi       CS       CFx       CFy       CFz       Cmx       Cmy       Cmz 
1         Wing                                       0.00100  10.00000   0.00000   0.09620   0.00169  -0.00341  -0.01549  -0.00341   0.09495   0.40035  -0.28337   0.05591 
2         Wing                                       0.00100  10.00000   0.00000   0.09620   0.00169   0.00341  -0.01549   0.00341   0.09495  -0.40035  -0.28336  -0.05591 
3         VTP                                        0.00100  10.00000   0.00000   0.00000  -0.00000  -0.00000  -0.00000  -0.00000   0.00000   0.00000  -0.00000  -0.00000 
4         HTP                                        0.00100  10.00000   0.00000   0.00511   0.00028   0.00000  -0.00061   0.00000   0.00508   0.00350  -0.04876   0.00043 

etc...

and from 3.25.0:

> Comp      Component-Name                             Mach       AoA      Beta       CL        CDi       CS       CFx       CFy       CFz       Cmx       Cmy       Cmz 
0                                                    0.00100  10.00000   0.00000   0.00000   0.00000   0.00000   0.00000   0.00000   0.00000   0.00000   0.00000   0.00000 
1         Wing                                       0.00100  10.00000   0.00000   0.09807   0.00113  -0.00258  -0.01592  -0.00258   0.09678   0.41015  -0.28849   0.05915 
2         Wing                                       0.00100  10.00000   0.00000   0.09808   0.00113   0.00258  -0.01592   0.00258   0.09678  -0.41015  -0.28849  -0.05915 
3         VTP                                        0.00100  10.00000   0.00000  -0.00000   0.00000  -0.00000   0.00000  -0.00000  -0.00000   0.00000   0.00000  -0.00000 
4         HTP                                        0.00100  10.00000   0.00000   0.00538   0.00042   0.00001  -0.00052   0.00001   0.00537   0.00367  -0.05134   0.00048 

etc...

ramcdona, 2021-08-31:
Was this thin-surface (VLM) or thick-surface (Panel)?  Was it using the legacy file format or the new experimental file format?

I think this behavior is expected as we move towards the new file format and since we've added support for load distributions on thick-surface files (only worked for VLM before).

I think this zero-component is used for non-lifting surfaces in thick-surface panel mode -- but I'm not 100% sure.

justingravett, 2021-09-21:
I asked Dave Kinney about this a little while ago and this was his response:
> "Basically the '0' is a catch all for anything that's not a wing. For the VLM we only have 2 types, and if you had a body it would have been listed."

As Rob suggested, this change was made in preparation of introducing the new *.vspgeom file format for the VSPAERO input geometry. For now, you can just ignore the blank component in the *.lod file but please let us know if it's leading to any problems elsewhere.

ramcdona, 2023-03-12:
If you're still having trouble, please post your question to the OpenVSP Google Group.

closed 2023-03-12 by ramcdona