OpenVSP/OpenVSP #199: Can't run VSPAERO on a mac with the new 3.25 version

mottiadar, 2021-09-09:
Hi

Using the new ver 3.25 on a mac.
Trying to run in either panels or VLM and getting an error:

ERROR 7: Could not open History file: /Users/motti/ADAR_Projects/Polarity/AerodynamicAssessments/.../file name.history
	File: /Users/runner/work/PrivateVSP/PrivateVSP/src/geom_core/VSPAEROMgr.cpp 	Line:2496
ERROR 7: Could not open Polar file: /Users/motti/ADAR_Projects/Polarity/AerodynamicAssessments/.../file name.polar
	File: /Users/runner/work/PrivateVSP/PrivateVSP/src/geom_core/VSPAEROMgr.cpp 	Line:2672
ERROR 7: Could not open Load file: /Users/motti/ADAR_Projects/Polarity/AerodynamicAssessments/.../file name.lod
	File: /Users/runner/work/PrivateVSP/PrivateVSP/src/geom_core/VSPAEROMgr.cpp 	Line:2908

(I removed some sub directories from the path and changed the file name with "file name")

Does anyone knows what can be done to fix it?

ramcdona, 2021-09-09:
Unfortunately, those errors don't tell us anything.  They are produced when OpenVSP fails to find files written by VSPAERO -- typically because it crashed.

I develop OpenVSP on a Mac and have not had this problem.  Can you tell us more about your situation?  Did a prior version work?  Are you on one of the new M1 Macs?  Anything else that might be pertinent?

mottiadar, 2021-09-09:
> ‫ב-9 בספט׳ 2021, בשעה 20:05, ‏‏Rob McDonald ***@***.***> כתב/ה:‬

Hi

I am using an intel based MacBook Pro. 
The case I was running is a simple wing case that I managed to run it on the previous version. 

Is there anything else I can send you that will help you find the problem?

Motti

ramcdona, 2021-09-09:
Is anything output to the console window at the bottom of the VSPAERO GUI?

You might try a default wing just to see if it is related to your model in some way.

Rob

mottiadar, 2021-09-09:
> ‫ב-9 בספט׳ 2021, בשעה 23:55, ‏‏Rob McDonald ***@***.***> כתב/ה:‬

I’ll try and let you know how it goes and what is output to the console windows. 

Thanks.

mottiadar, 2021-09-11:
Hi,

So I tried to run the default wing. I just started a new project and added the default wing. Didn’t change anything there and ran it. 

I got the same errors it the terminal.

And in the console window I get this message:
/Users/motti/Utils/OpenVSP-3.25.0-MacOS/vspaero -omp 4 Unnamed

And this is the only message I get. 
I don’t get the lines indicating the progress of the runs as I am used to see from previous versions.

Best regards,
Motti

ramcdona, 2021-09-15:
It looks like the 3.25.0 version of VSPAERO did not statically link the OpenMP libraries.

To work around this, you can install homebrew and then install gcc@10.  This will get you the OpenMP library and should hopefully resolve this issue.

I'll try to repair the build issue so this goes away with the next release.

closed 2021-09-15 by ramcdona