OpenVSP/OpenVSP #203: error Pixel OpenVSP

codeito, 2021-10-05:
Hello,
I met this issue when compiling OpenVSP (git clone --depth=1  https://github.com/OpenVSP/OpenVSP.git  repo), on CentOS-7.4 :

[ 54%] Building CXX object src/gui_and_draw/CMakeFiles/gui_and_draw.dir/CurveEditScreen.cpp.o
/stck/hmoussao/test/repo/src/gui_and_draw/CurveEditScreen.cpp: In member function ‘void CurveEditScreen::UpdateDrawObj()’:
/stck/hmoussao/test/repo/src/gui_and_draw/CurveEditScreen.cpp:750:38: error: ‘class VSPGUI::EditXSecWindow’ has no member named ‘pixels_per_unit’
     double point_size = m_XSecGlWin->pixels_per_unit() * edit_curve_xs->m_XSecPointSize.Get();
                                      ^~~~~~~~~~~~~~~
/stck/hmoussao/test/repo/src/gui_and_draw/CurveEditScreen.cpp:751:38: error: ‘class VSPGUI::EditXSecWindow’ has no member named ‘pixels_per_unit’
     double line_thick = m_XSecGlWin->pixels_per_unit() *edit_curve_xs->m_XSecLineThickness.Get();
                                      ^~~~~~~~~~~~~~~
make[2]: *** [src/gui_and_draw/CMakeFiles/gui_and_draw.dir/CurveEditScreen.cpp.o] Error 1
make[1]: *** [src/gui_and_draw/CMakeFiles/gui_and_draw.dir/all] Error 2
make: *** [all] Error 2

Thank you in advance, for help.

Best regards,

codeito, 2021-10-05:
The compiler is gcc.

ramcdona, 2021-10-05:
Thanks for this report.

The problem here is that the FLTK you're building against (likely bundled with CENTOS 7.4) is old and does not support high-dpi screens.

I can work on something to protect against this error down the road, but the ways to work around this is to either build the FLTK bundled with OpenVSP (and then build & link against it).  Or, to edit the lines where this appears and replace such that pixels_per_unit() would return 1.0.

There will be a few more similar occurrences.

codeito, 2021-10-21:
> Hello again,
>
> Here is the complement information :
>
> strings /usr/lib64/libdl.so.2 | grep -i glibc
> GLIBC_2.2.5
> GLIBC_2.3.3
> GLIBC_2.3.4
> GLIBC_PRIVATE
> __asprintf@@GLIBC_2.2.5
> free@@GLIBC_2.2.5
> strcpy@@GLIBC_2.2.5
> GLIBC_PRIVATE
> calloc@@GLIBC_2.2.5
> __dcgettext@@GLIBC_2.2.5
> GLIBC_2.3.4
> _rtld_global_ro@@GLIBC_PRIVATE
> dlopen@@GLIBC_2.2.5
> _dl_vsym@@GLIBC_PRIVATE
> _dl_addr@@GLIBC_PRIVATE
> _libc_intl_domainname@@GLIBC_2.2.5
> _dl_sym@@GLIBC_PRIVATE
> _dl_rtld_di_serinfo@@GLIBC_PRIVATE
> __environ@@GLIBC_2.2.5
> strerror@@GLIBC_2.2.5
> _rtld_global@@GLIBC_PRIVATE
> __cxa_finalize@@GLIBC_2.2.5
> GLIBC_2.3.3
> GLIBC_2.2.5
>
> Best regards,
>
> ---
> Cordialement,
>
> Houria Moussaoui
>
> DSI|OST
>
> 01 46 73 44 36
>
> ONERA - The French Aerospace Lab - Centre de Châtillon
> 29, avenue de la Division Leclerc - BP 72 - 92322 CHATILLON CEDEX

Hello,

I resolved this error.

Best regards,

-----
Cordialement,

Houria Moussaoui

DSI|OST

01 46 73 44 36

ONERA - The French Aerospace Lab - Centre de Châtillon
29, avenue de la Division Leclerc - BP 72 - 92322 CHATILLON CEDEX

Links:
------
[1] https://github.com/OpenVSP/OpenVSP/issues/203#issuecomment-934935231
[2] 
https://github.com/
[3] 
[4]

closed 2023-03-12 by ramcdona