OpenVSP/OpenVSP #211: 3.25.0 - POD geometry crashing system when change  from Wire to Shade.

neosoyo, 2021-11-09:
Hi there, 

i was trying to design a vehicle with 3 POD geometries, the software freezes and the notebook gives me the Black Screen Of Death when i change the  third POD from wire to shave in the surface menu over Geom Browser, this happen on every third POD (really weird):

[image]

i don't know if this is especific from my system yet, still need to test it over my desktop

System:

Pop!_OS 21.04 - kernel 5.13.0-7620-generic
AMD Ryzen 5 2500U APU notebook

justingravett, 2021-11-16:
I wasn't able to replicate it so I'll need a bit more info to be able to look into it. Can you share your VSP3 file?

ramcdona, 2022-01-02:
I'm closing this issue as not reproducible.  If this still happens for you with 3.26.0, please post an example file or exact steps to make it occur.

In general, the Google Group is a great place to ask questions about OpenVSP.

closed 2022-01-02 by ramcdona