OpenVSP/OpenVSP #212: missing vsp.py file

sreyese, 2021-11-11:
I'm trying to compile python openvsp but it's giving me this error, I couldn't find the file anywhere
 
>>> import openvsp
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/OpenVSP-main/src/python_api/packages/openvsp/openvsp/__init__.py", line 1, in <module>
    from .vsp import *

justingravett, 2021-11-16:
Have you taken a look at the [2020 Workshop Presentation](https://www.youtube.com/watch?v=oISkbtnyn7Q&ab_channel=OpenVSPVideos) about how to setup the Python API? You might see something that you missed when setting it up. Also to note is that the last few versions of OpenVSP mistakenly wrapped the API with Python 3.9. I believe this has been fixed in 3.25.0 but am not 100% sure. It will definitely be fixed by the next release, but just something to be aware of if that is causing your issue.

huyidao625, 2021-12-06:
The Python version must be the same as what OpenVSP was compiled with. For instance OpenVSP 3.21.2 Win64 requires Python 3.6-x64. If a different verison of Python is desired, the user must compile OpenVSP themselves.  [Python API Instructions](http://openvsp.org/api_docs/latest/)

iiwolf, 2022-01-14:
I'm having the same issue (Ubuntu 18.04)
```bash
Traceback (most recent call last):
  File "/home/ijw/projects/N221-004/OpenVSP/src/python_api/packages/../test.py", line 4, in <module>
    import openvsp as vsp
  File "/home/ijw/projects/N221-004/OpenVSP/src/python_api/packages/openvsp/openvsp/__init__.py", line 1, in <module>
    from .vsp import *
ModuleNotFoundError: No module named 'openvsp.vsp'
```

I have followed the instructions and tried with both Python 3.6 and 3.9 because it wasn't clear to me which 3.26.1 used.

iiwolf, 2022-06-30:
In my case this issue was the result of trying to `pip install` in the source directory instead of the install directory. In my case:

```/home/usr/openvsp/repo/src/python_api/packages```

Instead the install location:

```/usr/local/python```

luzpaz, 2024-10-18:
Should this be notated or fixed somewhere to avoid for posterity ?

ramcdona, 2024-10-22:
@luzpaz I can't actually identify what people are doing wrong from this thread.  It seems like a random group of people who are having generic trouble installing the Python package.  The first person thinks they're compiling Python. 

What would you fix?  What would you make note of?

luzpaz, 2024-10-22:
> What would you fix? What would you make note of?

Well if  more than 1 person was having this issue it may merit adding something to the documentation. Something like huyidao625 proposed: 

>>The Python version must be the same as what OpenVSP was compiled with. For instance OpenVSP 3.21.2 Win64 requires Python 3.6-x64. If a different verison of Python is desired, the user must compile OpenVSP themselves. [Python API Instructions](http://openvsp.org/api_docs/latest/)

But this may be too much of an edge-case and not worth it. Perhaps then closing the issue in that case ?

ramcdona, 2024-10-23:
Where would you put this information?  Do you think the person who needs it will find it there?

This is one of the core challenges of documenting a program like OpenVSP -- there is so much that could be done, but what is actually useful -- and how do you make it useful?

On top of all that, I'm terrible at Python.  I hate the beast and I certainly don't feel qualified to tell people how to manage their environment and Python setup.  I'm usually not sure what best practices for it are.

closed 2025-07-19 by ramcdona