OpenVSP/OpenVSP #218: Build Error: `(stackgeom_ptr) not declared`

acxz, 2022-01-19:
Error Log:
```
In file included from /usr/include/c++/11.1.0/cassert:44,
                 from /usr/include/glm/detail/func_exponential.inl:8,
                 from /usr/include/glm/exponential.hpp:110,
                 from /usr/include/glm/detail/func_geometric.inl:1,
                 from /usr/include/glm/geometric.hpp:116,
                 from /usr/include/glm/detail/func_matrix.inl:1,
                 from /usr/include/glm/matrix.hpp:161,
                 from /usr/include/glm/detail/type_mat2x2.inl:1,
                 from /usr/include/glm/detail/type_mat2x2.hpp:176,
                 from /usr/include/glm/ext/matrix_double2x2.hpp:5,
                 from /usr/include/glm/mat2x2.hpp:5,
                 from /usr/include/glm/glm.hpp:119,
                 from /home/acxz/vcs/git/github/acxz/pkgbuilds/openvsp/src/OpenVSP-OpenVSP_3.26.1/src/vsp_graphic/include/Display.h:7,
                 from /home/acxz/vcs/git/github/acxz/pkgbuilds/openvsp/src/OpenVSP-OpenVSP_3.26.1/src/gui_and_draw/ScreenBase.cpp:14:
/home/acxz/vcs/git/github/acxz/pkgbuilds/openvsp/src/OpenVSP-OpenVSP_3.26.1/src/gui_and_draw/ScreenBase.cpp: In member function ‘virtual void XSecScreen::GuiDeviceCallBack(GuiDevice*)’:
/home/acxz/vcs/git/github/acxz/pkgbuilds/openvsp/src/OpenVSP-OpenVSP_3.26.1/src/gui_and_draw/ScreenBase.cpp:2318:13: error: ‘stackgeom_ptr’ was not declared in this scope; did you mean ‘xsec_geom_ptr’?
 2318 |     assert( stackgeom_ptr );
      |             ^~~~~~~~~~~~~
```

System:
OS: archlinux
gcc: 11.1.0

ramcdona, 2022-01-21:
Thanks for this, I'll roll it into the next version.  This happened because I don't normally build in debug mode -- so asserts are always disabled.

In my experience, OpenVSP in debug mode is far too slow to be usable.  I make due with an RWDI build (release with debug info) for my development builds.

I only mention this because if this error is occurring for a build you are going to use -- it likely indicates that it is a Debug build -- and therefore will be extremely slow.

CMake defaults to a debug build unless you tell it otherwise.

acxz, 2022-03-19:
> Fixed in 3.27

closed 2022-03-19 by acxz