OpenVSP/OpenVSP #225: Cannot compile using recent STEPcode vesions

luke-allen, 2022-03-25:
Cannot compile OpenVSP from source using a recent external install of STEPcode. It fails when compiling util/CADutil.cpp, with the error:

```'class STEPattribute' has no member named 'Raw'```

The problem occurs for STEPcode versions after Aug 16, 2021 starting with [this commit](https://github.com/stepcode/stepcode/commit/5e1ad8f1c8aa164fd4f4f11a9bab2fece11144cb) in which the Raw() and  getEDesc() methods are removed. Downgrading STEPcode to the [previous commit](https://github.com/stepcode/stepcode/commit/799a51f340542178152ec4769a3fff61b3f4f43d) fixed the issue.

Can the required STEPcode version be specified? The latest official release (v0.8) on GitHub is from 2014, and the default branch when cloning is the "develop" branch.

ramcdona, 2022-03-29:
Thanks for raising this issue.

As you've noticed, STEPCode's development does not have a lot of structure.

OpenVSP bundles commit [28350d91294b](https://github.com/stepcode/stepcode/commit/28350d91294b) ca 2016 -- which includes some build fixes of our own, but has not been rolled into any more recent versions.

We (the OpenVSP devs) don't expect the user to have STEPCode installed on their machine -- or for them to download and build their own STEPCode.  I.e. it is an obscure enough library that we expect users to just use the version bundled with OpenVSP.

Recently the BRL-CAD team has been picking up STEPCode development and making progress (after a significant period of no activity in the project).  It is not clear exactly what their goals are, or when they will consider their work complete enough for another stable release of STEPCode.

I tried integrating a more recent version a while ago - but I prefer to avoid a change without specific purpose while the project is in such a state of flux.  When STEPCode stabilizes, I will likely try to update the OpenVSP bundled version -- and upstream any fixes required to make it work.

Are you a STEPCode developer?  Are you just trying to build OpenVSP on your own machine?  Do you have problems with the STEP files that you think are linked to the library?  In short, what are you trying to do?

If you are 'just' trying to build OpenVSP, I suggest you base your build process on the [GitHub Actions build processes](https://github.com/OpenVSP/OpenVSP/blob/main/.github/workflows/build.yml) we use to produce our 'official' builds.  Even if you do not follow it exactly, it conveys much intent.  For example, on Linux, I would suggest you build the libraries with a command very similar to:

cmake -DCMAKE_BUILD_TYPE=$BUILD_TYPE -DVSP_USE_SYSTEM_LIBXML2=true -DVSP_USE_SYSTEM_FLTK=true -DVSP_USE_SYSTEM_GLM=true -DVSP_USE_SYSTEM_GLEW=true -DVSP_USE_SYSTEM_CMINPACK=true ../Libraries

Which implies that the OS's package system should be used to supply libxml2, fltk, glm, glew, and cminpack.  These are relatively stable libraries that are available in most Linux distributions.

And all other libraries will be supplied by the versions bundled with OpenVSP.  These are relatively obscure libraries that can not reasonably be expected to be available:  Code-Eli, Triangle, Eigen (could reasonably be in first group), exprparse, libiges, stepcode.

On Mac and Windows, we expect you to use the bundled libraries for _all_ of the dependencies.

There are other libraries bundled in /src/external that we never give the option to use an 'external' version.  Most of these are just a few files (except AngelScript).

There is one important use case for using a 'non-bundled' version of a library -- if you are actively and simultaneously developing on the library and OpenVSP.  The most common example of this is Code-Eli.  On my own dev machine, I commonly point my OpenVSP build at an 'external' version of Code-Eli -- the version in a development directory on my machine.   In this way, I can simultaneously develop in Code-Eli and OpenVSP.

The same technique is often useful when debugging issues with external libraries (though I usually don't develop features in them).  I have done this for FLTK, STEPCode, LibIGES, Triangle, and a few others.

ramcdona, 2022-03-31:
All of the bundled library file names document the exact version -- either an official release version number, or a git commit SHA for a non-official version.  You can see a list [here](https://github.com/OpenVSP/OpenVSP/tree/main/Libraries).

Yours is a good reason to use an external build.  But you also sensibly want to start from what OpenVSP is using -- as you know it works with OpenVSP.  So, I would recommend exactly that -- set up an external build, start from the commit identified in the file name (my earlier message) and then work from there.

You might ask the BRL-CAD guys on the STEPCode mailing list (or their own mailing list) if they have any experience on your target machine.  I believe some of them have need to target some DOD HPC platforms.

There are a ton of changes in-between, it may be that one or two fix your problems.

Is there any way you can use a different compiler?  My (long ago) experience with Cray machines was that their compilers were often strange and grumpy.  I know they went through a period of ownership by SGI and Dell and probably others -- they aren't really the pure vector machines of yore.  If the sys admins on the machine have made clang or gcc available, you should consider it.

I presume you're already doing a no-graphics build -- it won't help for STEPCode, but it will greatly simplify building on a difficult architecture.

In an absolute worst case, if you can't get STEPCode to build on the machine, you could rip out all the STEP related stuff...  It would be a bit of work, but if you don't need to write STEP files, then ripping it out won't hurt anything else.

ramcdona, 2022-03-31:
I should have been explicit with that information in my first post.  By keeping that information in the file name, I don't have to keep a 3rd file up-to-date in any way.  It provides an easy way for me to make sure the exact version bundled can be traced.

Of course, the fact that I do this isn't documented anywhere...

Sounds like you're on the right track.  Best of luck.

closed 2022-03-31 by ramcdona